ghostop14/sparrow-wifi #1: ImportError: No module named 'PyQt5.QtChart'

fe8769a, 2017-11-23:
running sudo ./sparrow-wifi.py 

Traceback (most recent call last):
  File "./sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ImportError: No module named 'PyQt5.QtChart'

command lines 

sudo apt-get install python3-pip gpsd gpsd-clients
sudo pip3 install QScintilla PyQtChart gps3 dronekit manuf python-dateutil

have been done

ghostop14, 2017-11-24:
Hi fe8769a.  Hmmm, sounds like it's not finding PyQtChart.  Just to double-check if you do an apt search do you have python3-pyqt5 installed?  does 'pip3 search PyQtChart' show a version installed?  Just for good measure, what OS are you running on?

I checked on both my Ubuntu and Kali/Debian test systems and both report back like this:
'pip3 search pyqtchart'
PyQtChart (5.9.2)  - Python bindings for the Qt Charts library
  INSTALLED: 5.9
  LATEST:    5.9.2

fe8769a, 2017-11-25:
here is the result 
pip3 search pyqtchart
PyQtChart (5.9.2)  - Python bindings for the Qt Charts library
  INSTALLED: 5.9.1
  LATEST:    5.9.2
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

os is xubuntu 64 xfce

ghostop14, 2017-11-25:
We should be able to figure this out.  Sounds like python and QT5/QChart are crossed somewhere so we just have to find the difference.

So a couple things to try... do an 'ls -l /usr/bin/python3*' and see what comes back.  I've seen similar issues with multiple versions of python installed (python3 points one place and python3.5 points another so the pip3 didn't install the packages in the correct dist location).  Here's mine on Ubuntu 16.04:
lrwxrwxrwx 1 root root       9 May 27 07:35 /usr/bin/python3 -> python3.5
-rwxr-xr-x 2 root root 4456240 Sep 18 11:59 /usr/bin/python3.5
lrwxrwxrwx 1 root root      33 Sep 18 11:59 /usr/bin/python3.5-config -> x86_64-linux-gnu-python3.5-config
-rwxr-xr-x 2 root root 4456240 Sep 18 11:59 /usr/bin/python3.5m
lrwxrwxrwx 1 root root      34 Sep 18 11:59 /usr/bin/python3.5m-config -> x86_64-linux-gnu-python3.5m-conf
ig
lrwxrwxrwx 1 root root      16 Mar 23  2016 /usr/bin/python3-config -> python3.5-config
lrwxrwxrwx 1 root root      10 May 27 07:35 /usr/bin/python3m -> python3.5m
lrwxrwxrwx 1 root root      17 Mar 23  2016 /usr/bin/python3m-config -> python3.5m-config

The other thing to make troubleshooting quicker is you can just go 'python3' on the command-line and see if it'll even just let you do the import: 'from PyQt5.QtChart import QChart'.  It'll either work or fail right there.

I've also heard of errors if there's an issue in QtCore.  Try a 'pip3 search pyqt5'.  Here's what mine shows as installed:
PyQt5 (5.9.2)                     - Python bindings for the Qt cross platform UI and application toolkit
  INSTALLED: 5.9
  LATEST:    5.9.2
Qt.py (1.1.0b7)                   - Python 2 & 3 compatibility wrapper around all Qt bindings - PySide...
  INSTALLED: 1.0.0
  LATEST:    1.1.0b7

Searching with apt I also have these python3-pyqt5 packages installed there.
'i A python3-pyqt5                   - Python 3 bindings for Qt5                 
i A python3-pyqt5.qsci              - Python 3 bindings for QScintilla 2 with Qt
i A python3-pyqt5.qtsql             - Python 3 bindings for Qt5's SQL module    
i A python3-pyqt5.qtwebkit          - Python 3 bindings for Qt5's WebKit module

fe8769a, 2017-11-25:
That sound not easy to solve . here is the result of pip3 search pyqt5 :+1: 

vext.pyqt5 (0.5.21)               - Use system pyqt5 from a virtualenv
qutebrowser (1.0.3)               - A keyboard-driven, vim-like browser based
                                    on PyQt5.
Qt.py (1.1.0b7)                   - Python 2 & 3 compatibility wrapper around
                                    all Qt bindings - PySide, PySide2, PyQt4
                                    and PyQt5.
  INSTALLED: 1.0.0
  LATEST:    1.1.0b7
cheesemaker (0.3.8)               - A minimalistic image viewer using Python3
                                    and PyQt5
pyssb (0.1.1)                     - PyQT5 Site Specific Browser
python-pyqt5-vstructui (0.4.0)    - PyQt5 vstruct hex viewer widget.
ModernGL.PyQt5 (0.1.1)            - UNKNOWN
pytc-gui (1.2.2)                  - PyQt5 GUI for pytc API
quartz-browser (2.1.0)            - Fast Lightweight web browser written in
                                    PyQt5
pyqode.qt (2.10.0)                - Shim library that wraps PyQt5, PyQt4 and
                                    PySide
PyQt5 (5.9.2)                     - Python bindings for the Qt cross platform
                                    UI and application toolkit
  INSTALLED: 5.9.1
  LATEST:    5.9.2
Sonance-Music-Player (0.1.Alpha)  - This is a music player using PyQt5
pyblish-qml (1.6.0)               - Frontend for Pyblish written in PyQt5/QML
tws-async (0.5.7)                 - Use the Interactive Brokers API (IBAPI)
                                    asynchonouslywith asyncio or PyQt5
specguiutils (0.5)                - Library to provide PyQt5 widgets to
                                    display spec file information read using
                                    spec2nexus.spec file library
qutepart (3.1.0)                  - Code editor component for PyQt5
AnyQt (0.0.8)                     - PyQt4/PyQt5 compatibility layer.
python-pyqt5-hexview (0.4.3)      - PyQt5 hex viewer widget.
xcirculardichro (0.7)             - Library to provide PyQt5 widgets to
                                    display spec file information read using
                                    spec2nexus.spec file library
yxspkg_songzgif (1.0.1)           - A GUI to make gif based on pyqt5
fgmk (0.7.3)                      - A PyQt5 Maker to generate a RPG Javascript
                                    game.
python-qdatamatrix (0.1.16)       - A PyQt4/PyQt5 widget for viewing and
                                    editing a DataMatrix object
QtPy (1.3.1)                      - Provides an abstraction layer on top of
                                    the various Qt bindings (PyQt5, PyQt4 and
                                    PySide) and additional custom QWidgets.
python-qt5 (0.3.0)                - PyQt5
world2d (0.2.0)                   - A widget for PyQt5 implementing a 2d world
                                    space to draw in.
pyqt5-macos-built (5.5.0)         - This package installs a prebuilt version
                                    of PyQt5 for Mac OS
qt_backport (0.1.3)               - Makes PySide/PyQt4 code work with Qt5
                                    (using PyQt5)
QtStuff (0.4)                     - Compatibility layer and helper classes for
                                    PySide/PyQt4/PyQt5
pynoder (1.0.3)                   - A Python and PyQt5 package for creating a
                                    node-based editor.
qroundprogressbar (1.0.0)         - PyQt5 port of C++ circular progress bar
                                    widget library
hackedit (1.0a2)                  - The hackable IDE, built with Python3,
                                    PyQt5 and pyQode
logReader (0.7.6)                 - PyQt5 programm for parse and reading any
                                    log file.
SiQt (0.2.0a1)                    - A compatibility library for PyQt4, PyQt5
                                    and PySide
PyQtX (0.1.2)                     - Mimics the structure of PyQt5 but provides
                                    PyQt4 as a fallback solution.
pyqt5-installer (5.7.0)           - This package installs an script to install
                                    PyQt5
pyqt5-tools (5.9.0.1.2)           - Tools to supplement the official PyQt5
                                    wheels
pyqt5reactor (0.0.1)              - 
pyqtspinner (0.1.1)               - A waiting spinner for PyQt5
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

ghostop14, 2017-11-26:
Looks like you have the same things there.  Let's try the import in a couple of python variations and see if we can figure out if it maybe didn't put it in all the distro's.

From a command-line type 'python3.5'.  You should get a python interactive command prompt.  type 'from PyQt5.QtChart import QChart' and see if it throws the same error (On Ubuntu 16.04 python 3.5 is the version it's installed but it's possible the python3 symlink may not be correct, this'll be a quick test).

fe8769a, 2017-11-28:
here is the result

Python 3.5.2 (default, Sep 14 2017, 22:51:06) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyQt5.QtChart import QChart
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'PyQt5.QtChart'
>>>

ghostop14, 2017-11-28:
Lets try forcing a reset...

sudo apt-get install --reinstall python3-pyqt5
pip3 install --upgrade --force-reinstall PyQtChart

fe8769a, 2017-11-28:
sudo apt-get install --reinstall python3-pyqt5
pip3 install --upgrade --force-reinstall PyQtChart

worked fine except error message on pip3 version said it is 8.11 sould be 9.0.1 

and still not working 

sudo ./sparrow-wifi.py
Traceback (most recent call last):
  File "./sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ImportError: No module named 'PyQt5.QtChart'

very strange

ghostop14, 2017-11-29:
This fixed a QT5 widgets issue in another project... see if this helps:

sudo apt-get install qt5-default

ghostop14, 2017-11-29:
Tried it on a fresh Ubuntu 16.04.3 install and there were a couple items that seem unrelated that I added to the install instructions.  See if these help at all too.

sudo apt-get install python3-tk
sudo pip3 install numpy matplotlib

ghostop14, 2017-11-29:
I'm with you I hate having to rebuild or try a clean version.  I'd prefer to fix whatever's going on.  Just to see if it's something unique to that system, it may be worth rather than reloading, trying a fresh XUbuntu in a virtual machine (VMWare, virtual box, etc.).  Do a quick base VM, nothing fancy, install the prerequisites and see if it works there.  That way you don't have to disrupt that system at all.

fe8769a, 2017-11-29:
Will do if I have some time left ..

aimdev, 2018-09-08:
Debian Stretch
The only way I managed to get sparrow-wifi to work was to carry out the following.
remove all python3 using apt
comment out references to stretch in /etc/apt/sources.list
add the sid repository
do an update
DO NOT DO AN UPGRADE!!!!
installed python3.5, pip etc
install all libraries
run program, it works
edit  sources.list, uncomment stretch references, comment/remove sid reference
do an update (however all upgrades to python will now be invalid as debian's stretch release is behind the required release, the people on github mu-editor have had similiar issues) 
your python3 executable is now python3.5, so a symbolic link will be required.

hecoupy, 2018-10-07:
> […](#)

Have you solved this issue? I have  the same issue with you.

Cody-Chen, 2018-11-08:
Have you solved this issue? I have the same issue with you.
python3.5 + PyQT5 ，have install the pyqtchart via pip3.

PyQT5 demo working well . I use `pip3 list` to check those thing. everything is fine. 

but get  `ImportError: No module named 'PyQt5.QtChart`

ghostop14, 2018-11-08:
I can't reproduce the problem, so my suspicion is it's a fundamental Python issue on that system.  I've tested it on Ubuntu 16, Kali linux, and a straight Debian 9, and everything seems to work fine from a standard OS  installation.

Cody-Chen, 2018-11-09:
After a day and night. I have solve my issue.
I love to use PyQt5 and make some toys for myself. I use Python27 and  Python35 in my Ubuntu16.
I install PyQt5  via `apt-get` before. and The version is old "PyQt5 5.5".
How to check the version of PyQt5 ? 
```
$python2 
>>>from PyQt5 import QtCore ; 
>>>print(QtCore.PYQT_VERSION_STR).
```
```
$python3 
>>>from PyQt5 import QtCore ; 
>>>print(QtCore.PYQT_VERSION_STR).
```
I get both version string is 5.5.0. But I have check the pip3 list ,It has `PyQt5 5.11`. It means something is going wrong.
Now I have uninstall all PyQt5 on my system. 
run :  `sudo find / -type d -iname "PyQt5*"`
Looking for all about PyQt5 and `rm -r` them. 

Following is :  Install PyQT5/SIP for Py27 and Py35. Thay are all supported.

#For Py35 or later version:
pip3 install is avaliable.
Just run pip3 install PyQtChart. It will auto install "PyQt5" and "PyQt5-sip".
After installed. could check version with:
`from PyQt5 import QtCore`
`print(QtCore.PYQT_VERSION_STR)`
you will get a string version >5.11. That is a very new version.It is support the PyQtChart.
The PyQt5 version of apt-get is very old. about 5.5. It not support QtChart.

#For Py27: (no pip whl support. manual build sip and pyqt5, and not support PyQtChart)

1. build SIP: 2 part  "python27-sip" and "PyQt5-sip".
For Py27. need to build SIP and PyQt5-sip. 
https://www.riverbankcomputing.com/software/sip/download; (For now :2018-11 sip-4.19.13.tar)
-----------------------------
"python27-sip"
>python configure.py
>make
>make install
------------------------
"PyQt5-sip".
make sure "python" is python2.7, We just install for python27
>python configure.py --sip-module PyQt5.sip --no-tools
>make
>make install
------------------------
2. build PyQT5 for Py27
download PyQT5 shouce code.https://www.riverbankcomputing.com/software/pyqt/download5 
(For now :2018-11 PyQt5_gpl-5.11.3.zip)

following step will check "python27-sip" and system sip version first. manual chaeck run:`sip -V.` 
>python configure.py 
>make
>make install

ghostop14, 2018-11-09:
Excellent!    It did seem like a local config issue.  Hopefully that solution helps others.

closed 2018-11-09 by ghostop14

zacwang89, 2019-10-11:
I use pip to install PyQt5 for the system version of Python, which turns out it also doesn't include QtChart. python3 -m pip install PyQt5==5.13.1
After that, I use python3 -m pip install PyQtChart, then system Python can import QtChart, so I think pip install default method won't install QtChart.

wgr-vremsoft, 2020-03-28:
> I use pip to install PyQt5 for Python3.6, which turns out it also doesn't include QtChart. python3 -m pip install PyQt5==5.13.1
> After that, I use python3 -m pip install PyQtChart, then Python can import QtChart, so I think pip install default method won't install QtChart.

Thanks mate, you're my hero ^^

pavan555, 2020-06-16:
this worked for me pip install PyQtChart==5.13.1

gconelhero, 2020-08-26:
Digite no terminal:
`sudo apt-get install python3-PyQt5.QtChart`
Resolvido!
Debian 10 Buster

Jhongesell, 2020-08-28:
Me funcionó , ya tenía instalado el PyQt5; solamente faltaba esta librería, gracias por la ayuda @pavan555 .

GreenTeaHua, 2020-10-10:
pip install PyQtChart
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Collecting PyQtChart
  Downloading https://pypi.tuna.tsinghua.edu.cn/packages/fc/12/7effbf8f283642b7839e470334a726ecb4897e666d129e8372b641180076/PyQtChart-5.15.1-5.15.1-cp35.cp36.cp37.cp38.cp39-none-win_amd64.whl (927 kB)
     |████████████████████████████████| 927 kB 2.2 MB/s
Requirement already satisfied: PyQt5>=5.15 in e:\app\anaconda3\lib\site-packages (from PyQtChart) (5.15.1)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in e:\app\anaconda3\lib\site-packages (from PyQtChart) (12.8.1)
Installing collected packages: PyQtChart
Successfully installed PyQtChart-5.15.1
its ok.