ghostop14/sparrow-wifi #2: Pi remote agent error

JoeTester1965, 2017-12-05:
Followed instructions.

1. visit myIP:\\8020

2. Server: 192.168.1.111 - - [05/Dec/2017 14:10:30] "GET / HTTP/1.1" 404 

3. Browser : Bad request

ghostop14, 2017-12-05:
I still have yet to really document the URLs, but if you search in sparrowwifiagent.py there's an isValidGetURL() and isValidPostURL() set of functions.  There's no response to a GET / request, but rather they're grouped by functionality.

This query should work for you from a command-line with the agent running:
curl http://127.0.0.1:8020/wireless/interfaces

You should get json back that looks something like:
{"interfaces": ["wlp3s0"]}

Then if you use that name in a follow-up query it will run the scan and return the results as JSON:
curl http://127.0.0.1:8020/wireless/networks/wlp3s0

Those are the two basic calls the UI uses to "run a remote scan".  The wirelessengine.py file has a class that can take the JSON and populate a wireless network object if you're rolling your own.

luiz-orlandi, 2018-12-07:
I'm having the same issue, but when I run the "http://127.0.0.1:8020/wireless/networks/wlp3s0" the send me back a bunch of info...

`{"errCode": 0, "errString": "", "gps": {"latitude": 0.0, "longitude": 0.0, "altitude": 0.0, "speed": 0.0}, "networks": [{"type": "wifi-ap", "macAddr": "c4:13:e2:39:d5:e4", "ssid": "SWN_CORP", "mode": "AP", "security": "IEEE 802.1X FT/IEEE 802.1X", "privacy": "CCMP", "cipher": "CCMP", "frequency": 5580, "channel": 116, "secondaryChannel": 120, "secondaryChannelLocation": "above", "thirdChannel": 118, "signal": -46, "strongestsignal": -46, "bandwidth": 40, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "ee:9f:db:89:8e:6c", "ssid": "", "mode": "AP", "security": "PSK", "privacy": "CCMP", "cipher": "CCMP", "frequency": 2412, "channel": 1, "secondaryChannel": 5, "secondaryChannelLocation": "above", "thirdChannel": 0, "signal": -76, "strongestsignal": -76, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "de:9f:db:89:8e:6c", "ssid": "redeescola", "mode": "AP", "security": "PSK", "privacy": "CCMP", "cipher": "CCMP", "frequency": 2412, "channel": 1, "secondaryChannel": 5, "secondaryChannelLocation": "above", "thirdChannel": 0, "signal": -73, "strongestsignal": -73, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "2e:a4:3c:89:4d:36", "ssid": "ROBO-D", "mode": "AP", "security": "PSK", "privacy": "CCMP/TKIP", "cipher": "CCMP/TKIP", "frequency": 2412, "channel": 1, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -77, "strongestsignal": -77, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "32:a4:3c:89:4c:dd", "ssid": "ROBO-F", "mode": "AP", "security": "PSK", "privacy": "CCMP/TKIP", "cipher": "CCMP/TKIP", "frequency": 2412, "channel": 1, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -72, "strongestsignal": -72, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "12:18:d6:a9:60:3e", "ssid": "PRUDENT BRAZIL PRIVADO", "mode": "AP", "security": "PSK", "privacy": "CCMP", "cipher": "CCMP", "frequency": 2412, "channel": 1, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -48, "strongestsignal": -48, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "56:be:f7:46:6d:19", "ssid": "#NET-CLARO-WIFI", "mode": "AP", "security": "Open", "privacy": "None", "cipher": "", "frequency": 2412, "channel": 1, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -53, "strongestsignal": -53, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "2e:a4:3c:89:4c:dd", "ssid": "ROBO-D", "mode": "AP", "security": "PSK", "privacy": "CCMP/TKIP", "cipher": "CCMP/TKIP", "frequency": 2412, "channel": 1, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -61, "strongestsignal": -61, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}, {"type": "wifi-ap", "macAddr": "58:8b:f3:14:2c:95", "ssid": "LIVE TIM", "mode": "AP", "security": "PSK", "privacy": "CCMP/TKIP", "cipher": "CCMP/TKIP", "frequency": 2427, "channel": 4, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -79, "strongestsignal": -79, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0 [... cut ...] requency": 5745, "channel": 149, "secondaryChannel": 0, "secondaryChannelLocation": "no", "thirdChannel": 0, "signal": -90, "strongestsignal": -90, "bandwidth": 20, "firstseen": "2018-12-07 13:52:50.990816", "lastseen": "2018-12-07 13:52:50.990816", "lat": "0.0", "lon": "0.0", "alt": "0.0", "speed": "0.0", "gpsvalid": "True", "strongestlat": "0.0", "strongestlon": "0.0", "strongestalt": "0.0", "strongestspeed": "0.0", "strongestgpsvalid": "False"}]}`

 How can I get the GUI up ?!

Regards;

Luiz