ghostop14/sparrow-wifi #3: UHD Support

shepardac, 2017-12-27:
Do you see any addition of UHD support for SDR's in the future?

nlovlyn, 2018-01-19:
I second this request. I would love to use your software with my USRP B200.

ghostop14, 2018-01-28:
Hi folks, I have a B200 too I could use, but I'm not so great with the FPGA side of things to make it all come together.  For Sparrow I use the hackrf_sweep command-line tool to handle all the frequency analysis.  It basically streams me FFT's in given bucket sizes and handles all the radio channel changes and FFT work on the hardware.  If we had a tool that output something similar for the USRPs it wouldn't be much work at all to integrate it.

mwk088, 2019-09-11:
Has there been any movement on this?  I would like to funnel IQ data from a USRP X310 or N210.  What would the effort be on getting something like this completed?

ghostop14, 2019-09-11:
Hi mwk088, I don't have any plans to incorporate USRP streams into sparrow at this time.  Hackrf_sweep is by far the best performing solution at the moment, since it does all the work for this in hardware.  Have you thought about doing it over in GNU Radio?

mwk088, 2019-09-11:
Hi ghostop14,

Thanks for the quick response.  I am using GNURADIO 3.8 with python3, and I was hoping to find a ready-made wifi analyzer and your Sparrow Wifi Analyzer gui featured on the main page looks awesome.  I want to utilize the USRP SDR with the wifi analyzer.

ghostop14, 2019-09-11:
hackrf_sweep is the most efficient for the broadband spectrum view.   It handles the FFT and channel hopping to work through the spectrum in the FPGA.  Trying to pull the samples at 60-100 MSPS, do the FFT's and channel hop would just crush a python app.  You can see the difference between doing it in SW versus HW using qspectrumanalyzer.  Try it with hackrf_sweep then an airspy or rtl-sdr.  The frames/sec is incredibly slow.

mwk088, 2019-09-11:
Thanks for the suggestion of hackrf_sweep, however, will it work with a USRP?

ghostop14, 2019-09-11:
Nope.  Hackrf_specific.  I would bet though you could write a usrp-equivilent.  Probably need an FPGA image and stand-alone C++ program that matches the output of hackrf.  That would certainly do the trick.  If that existed, I could tweak sparrow to support that.  I'm just not an FPGA guy by any means to implement it.

alphafox02, 2020-12-14:
Just a thought on this - if you really want to use the x310 to scan for WiFi you can use gr-80211 and it’s included script that also brings up wireshark. Be interesting to see if there’s some way to tie this into sparrow. 

DragonOS Focal GR-IEEE802.11 w/ Ettus X310 + TwinRX80 (GNU Radio, Wireshark, X310) https://youtu.be/rOtnq7H2-5o

ghostop14, 2020-12-14:
I'd love to have an X310 here (I do a good bit of signals work) to test with.  One thought in that flowgraph is that you could add a block that writes any detected info to a json or csv text file in the format sparrow can read, then just import/open it.  Or you could take the input and have a block that responds to the HTTP requests the same way the agent does an pipe it in that way.  Wifi decoding on the CPU is pretty intensive, so I like the idea of having them on separate systems.  Overall I think you'll get better results with the wifi cards, unless you're using unusual wifi modes that wifi cards don't generally support.  And that flowgraph I don't think I've ever seen it go above 20 MHz, which is pretty narrow by today's wifi standards.

alphafox02, 2020-12-14:
I still have the x310 here with 2 twinrx 80 cards in it but I don’t know enough about it to figure out how to use both cards at the same time. Like I thought I’d see double the available bandwidth in gqrx but so far that’s not the case. I’ll try again. I’ll also try to widen the bandwidth in the flow graph and see what happens if I put it on a channel that’s 40mhz wide or more. 

I wish I could add a block like you mentioned. I’ll probably have to wait till I get some more knowledge under my belt.