ghostop14/sparrow-wifi #4: PyQt5.QtChart ERROR

EastwindZL, 2018-01-29:
OS:Kali2017.3

error:
root@:~/sparrow-wifi# ./sparrow-wifi.py
Traceback (most recent call last):
  File "./sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ModuleNotFoundError: No module named 'PyQt5.QtChart'

Hope to help me!!!thanks!

closed 2018-03-10 by EastwindZL

hammoumorsli, 2019-04-28:
**pip install PyQtChart**

ghostop14, 2019-04-29:
Depending on how your python behaves, since sparrow is in Python3 you may need to 'pip3 install PyQtChart'  if the 'pip install' only installs it for Python2.

trumptarder, 2019-11-09:
```
git clone https://github.com/ghostop14/sparrow-wifi
cd sparrow-wifi
virtualenv --python=python3 $HOME/sparrow
source $HOME/sparrow/bin/activate
pip3 install gps3 python-dateutil requests pyqt5 pyqtchart numpy matplotlib
sudo python3 sparrow-wifi.py

  File "sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ModuleNotFoundError: No module named 'PyQt5.QtChart'
```

> pip3 install PyQtChart                                                                                                           
> Requirement already satisfied: PyQtChart in /home/****/sparrow/lib/python3.7/site-packages (5.13.1)
> Requirement already satisfied: PyQt5>=5.13 in /home/****/sparrow/lib/python3.7/site-packages (from PyQtChart) (5.13.2)
> Requirement already satisfied: PyQt5_sip<13,>=4.19.19 in /home/****/sparrow/lib/python3.7/site-packages (from PyQt5>=5.13->PyQtChart) (12.7.0)
>

ghostop14, 2019-11-11:
I've seen issues with PyQtChart on some OS's, like Raspbian where it's not included in the repo.  There's a note in the Raspberry Pi section and a script (rpi.setup_prerequisites.sh) that may provide help if QtChart isn't included in your OS build.  Running on Kali noted by luckyfishgeek should work.  I run the latest Kali in a VM with it (Kali 2019 kept up to date) and it runs fine with the standard build instructions.

maksokami, 2020-02-10:
Trying to install on Kali with the default instructions and running into the same issue (tried both with and without virtualenv)

ghostop14, 2020-02-11:
I actually just stumbled on the solution in the past week.  If you do a "sudo pip3 install --upgrade pip" before doing the pip3 installs, it'll work.  Something in PyQtChart must be using a new pip feature or something.  Try that and let me know if it works.

mmuncan, 2020-05-02:
hi!
ghostop14: no unfortunately this dint work too... :-/

i m running latest kali (2020.2 i guess ;))

ghostop14, 2020-05-04:
QtChart has been a bit of a pain with repo versions and pip upgrades.  I'm on Ubuntu 18.04.3 LTS and I recently had to add /usr/local/lib/python3.6/dist-packages/PyQt5/Qt/lib to my LIBRARY_PATH to get it to pick up the correct QtChart library.  Seemed like there were a number scattered around the system.  But whey you pip install, the correct library version may only be in the dist-packages path.  See if that helps at all?

UmutErgunes, 2020-08-18:
hi!

I also use latest kali linux 2020.2. I've tried to follow your readme but when I try to run ./sparrow-wifi.py it gives me same problem.
I've tried to "pip3 install PyQtChart" and it gives me this:

_Requirement already satisfied: PyQtChart in /usr/local/lib/python3.8/dist-packages (5.15.0)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in /usr/local/lib/python3.8/dist-packages (from PyQtChart) (12.8.0)
Requirement already satisfied: PyQt5>=5.15 in /usr/lib/python3/dist-packages (from PyQtChart) (5.15.0)_

Could someone help me please?

ghostop14, 2020-08-19:
I'll have some time next week to spin up a latest kali and check it out.  Standby, I'll see what's up.

UmutErgunes, 2020-08-19:
Okay thnx