ghostop14/sparrow-wifi #8: build of dronekit fails; can I run sparrow-wifi without it? Don't need drone support right now...

jkbecker, 2018-06-28:
Hi,

I tried installing the dependencies for sparrow-wifi, but installing dronekit fails [(see log)](https://gist.github.com/jkbecker/0391db37ebf6ff290d1ebac9fa553848#file-install-log-L179).

Looking at the dronekit website it seems that dronekit currently does not build on python 3, maybe due to some [MAVProxy incompatibility](https://github.com/ArduPilot/MAVProxy/pull/420)?

Is there a trick to get sparrow running without the drone component in the meantime? Any help appreciated.

ghostop14, 2018-06-29:
I just pushed out a quick code update so the agent can run without dronekit installed.  Give it a shot and let me know if that did the trick.

jkbecker, 2018-06-29:
Cheers! I'll try that out in a bit and report back.

jkbecker, 2018-06-29:
Yes works great! Thanks for the quick response & fix!

closed 2018-06-29 by jkbecker