ghostop14/sparrow-wifi #9: Could not find a version that satisfies the requirement QScintilla

DTeCH-X, 2018-06-29:
root@Microknoppix:/home/knoppix# sudo pip3 install QScintilla PyQtChart gps3 dronekit manuf python-dateutil numpy matplotlib
Collecting QScintilla
  _Could not find a version that satisfies the requirement **QScintilla** (from versions: )
No matching distribution found for **QScintilla**_

Running on **Knoppix 8.2 64bit**

ghostop14, 2018-06-29:
Give it a shot without qscintilla.  I ran into that on one of my systems that was a light kernel but not others.  I think it may work (I dont explicitly use it, I hink another component may) without it but give it a shot.

ghostop14, 2019-09-24:
You can probably get away without the dronekit piece if that's an issue, but yeah, you may need to see if there's a build-from-source for Knoppix for QScintilla.  A couple years back I ran into similar issues on a Raspberry Pi that prevented me from running the GUI on it, but their repo caught up and it works now.  Plan B could always be to run the agent on the Knoppix platform and connect to it with the UI from a different system.  All of the functionality in the UI when running local can also be used via the remote agent.