ghostop14/sparrow-wifi #10: hcitool is missing

ex47, 2018-07-16:
```
ex47@k ~/sparrow-wifi-master $ sudo ./sparrow-wifi.py
Password: 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Traceback (most recent call last):
  File "./sparrow-wifi.py", line 3670, in <module>
    mainWin = mainWindow()
  File "./sparrow-wifi.py", line 668, in __init__
    self.checkForBluetooth()
  File "./sparrow-wifi.py", line 635, in checkForBluetooth
    numBtAdapters = len(SparrowBluetooth.getBluetoothInterfaces())
  File "/home/ex47/sparrow-wifi-master/sparrowbluetooth.py", line 970, in getBluetoothInterfaces
    result = subprocess.run(['hcitool', 'dev'], stdout=subprocess.PIPE,stderr=subprocess.DEVNULL)
  File "/usr/lib/python3.6/subprocess.py", line 403, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'hcitool': 'hcitool'
```
And it seems like it's missing for good now:
https://github.com/chrippa/ds4drv/issues/123
https://github.com/peplin/pygatt/issues/112

ghostop14, 2018-07-16:
Hi ext47, hcitool on the Debian variants including Ubuntu is part of the bluez bluetooth tools package.  See if an "apt install bluez" helps?

ex47, 2018-07-16:
I'm on Solus. I have bluez 5.48-22 installed. Apparently it comes without hcitool. 

As I understand hcitool is for bluetooth scanning. I don't really need to scan bluetooth, just wifi. Can I disable bluetooth part?

ghostop14, 2018-07-16:
Just pushed an update to the bluetooth for you.  If an exception gets thrown running hcitool it'll gracefully print a message to stdout and return a blank device list.  That should let you run without bluetooth.  Let me know if that does the trick.