ghostop14/sparrow-wifi #16: I donot know why?

hecoupy, 2018-10-04:
Traceback (most recent call last):
  File "./sparrow-wifi.py", line 33, in <module>
    from PyQt5.QtWidgets import QApplication, QMainWindow,  QDesktopWidget, QGraphicsSimpleTextItem, QFrame, QGraphicsView
ImportError: /usr/local/lib/python3.5/dist-packages/PyQt5/QtCore.so: undefined symbol: PySlice_AdjustIndices

ghostop14, 2018-10-04:
Hi hecoupy, seems like something is wrong with the QT5 installation that QtCore doesn't contain a function.  What OS/platform are you running on?

hecoupy, 2018-10-04:
ubuntu 16.04

hecoupy, 2018-10-04:
> Hi hecoupy, seems like something is wrong with the QT5 installation that QtCore doesn't contain a function. What OS/platform are you running on?

ubuntu 16.04

ghostop14, 2018-10-04:
That's what I generally run it on too so it should work.  I think the import error path lends a clue....  /usr/local/lib/python3.5/dist-packages/PyQt5/QtCore.so.

Sometimes python updates can cause issues with packages.  Maybe try a "sudo apt-get install --reinstall python3-pyqt5".  Just looks like something is not right with the Python QT5 version.  Also have you run apt-get update && apt-get upgrade at all to make sure you have all the latest packages?

stryngs, 2021-05-20:
Try a virtual environment to narrow it down

Let us get risque here...  As root, do:

python3 -m venv env
source env/bin/activate
python3 ./sparrow-wifi.py

Keep pip3 installing until all requirements are met.

Bob's your uncle.