ghostop14/sparrow-wifi #18: Error at sparrowwifiagent dateutil

luiz-orlandi, 2018-12-07:
Hi guys !!
 I'm getting an error when tries to run the "sudo ./sparrowwifiagent.py"

Traceback (most recent call last):
  File "./sparrowwifiagent.py", line 32, in <module>
    from dateutil import parser
ModuleNotFoundError: No module named 'dateutil'

I've installed the Dateutils, but the scritp dont run.

Can you help me ?!

Regards;
Luiz

ghostop14, 2018-12-07:
Sounds like you're missing a python module.  Dateutil isn't present in your python install.  The easiest way to fix it will be 'sudo pip3 install python-dateutil'.  See if that does the trick. (if your OS can't find pip3, you may need to 'sudo apt install python3-pip' first)

ghostop14, 2018-12-07:
Sounds right.  It's not a UI on the web server but rather a set of API calls.  The UI calls them, but you can also call them directly.  Each API call will return a JSON result.  So /wireless/interfaces will return a JSON list of wireless interfaces, /wireless/interfaces/<interface> will scan using the specified interface, etc.  

Were you looking for a UI on the web server?

luiz-orlandi, 2018-12-07:
Sry !!
I was executing the wrong script.... 

Regards;

Luiz

ghostop14, 2021-05-21:
Thanks folks!

closed 2021-05-21 by ghostop14