ghostop14/sparrow-wifi #20: Add instructions to README for installing dependencies in a virtualenv

C0rn3j, 2019-02-03:
Would love if the instructions were in the readme as I wanted to run this project on Arch Linux but tracking down the dependencies was annoying.
```
git clone https://github.com/ghostop14/sparrow-wifi
cd sparrow-wifi
virtualenv ~/sparrow
source ~/sparrow/bin/activate
pip install gps3 python-dateutil requests pyqt5 pyqtchart numpy matplotlib
sudo python sparrow-wifi.py
```

q2dg, 2019-03-05:
Well, it seems this project is pretty abandoned, sadly ...

ghostop14, 2019-03-05:
Good morning q2dg, this project is definitely not abandoned.  I'm actually heading out for a project this morning where I use it in support of my day job.  I don't typically work in that mode, so if you have some instructions you've found work for you running it in a virtualenv, I'm happy to incorporate it into the readme.  Just write it up in a way that is readme-ready and I'll add it to it.  It wasn't clear from what was pasted by C0rn3j if that was everything in the process to get it working (i.e. after the commands pasted above, did it work? or is that what was attempted and it didn't work?).

C0rn3j, 2019-03-05:
Been a while but I believe it worked afterwards. This should be added as a general guide to run it under virtualenv, it's not Arch specific.

ghostop14, 2019-03-05:
Tested it real quick... made a few minor edits to make sure it searched for python3 and was copy/scriptable:

git clone https://github.com/ghostop14/sparrow-wifi
cd sparrow-wifi
virtualenv --python=python3 $HOME/sparrow
source $HOME/sparrow/bin/activate
pip3 install gps3 python-dateutil requests pyqt5 pyqtchart numpy matplotlib
sudo python3 sparrow-wifi.py

I'll put it in the README momentarily.  Thanks!

closed 2019-03-05 by ghostop14