ghostop14/sparrow-wifi #21: qtchart error

awesome14, 2019-06-27:
I get this error: File "sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ImportError: No module named 'PyQt5.QtChart'

I'm using debian stretch. I followed the read me. I'm running sparrow-wifi.py as root.

ghostop14, 2019-06-29:
Sounds like something is up with your pyqtchart install.  Give this a shot and see what it says: "pip3 search pyqtchart".

You should get a response something like:
PyQtChart (5.12)  - Python bindings for the Qt Charts library
  INSTALLED: 5.12 (latest)

If it's installed correctly, from a command line you should be able to just simply do "python3" then type "from PyQt5.QtChart import QChart", or even more generically (just for testing) try "import PyQt5.QtChart".  If you get any errors, sounds like some local install troubleshooting.

awesome14, 2019-07-18:
From the first command I get: "PyQtChart (5.13.0)  - Python bindings for the Qt Charts library"

From the second commands I get: Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'PyQt5.QtChart'

I can look into it, but I thought you should be aware of it. If you  have a one-liner to solve it, let me know.

JoshOxford, 2019-08-14:
I assume you are using a virtualenv if so:

The issue is likely due to running the script as root the the virtualenv environment variables, aliases, functions, etc aren't being carried over.

Fix: When you run the script tell it to use the path to the virtualenv e.g. sudo /home/user/sparrow/bin/python3 sparrow-wifi.py

rikosintie, 2019-12-26:
Thanks Josh, This solved it for me:
Fix: When you run the script tell it to use the path to the virtualenv e.g. sudo /home/user/sparrow/bin/python3 sparrow-wifi.py

stryngs, 2021-05-20:
> Thanks Josh, This solved it for me:
> Fix: When you run the script tell it to use the path to the virtualenv e.g. sudo /home/user/sparrow/bin/python3 sparrow-wifi.py

Close this =)