ghostop14/sparrow-wifi #23: Outdated blue_hydra source

ZeroChaos-, 2019-11-07:
Hi!  I read an article about your tool this morning, I am excited to try it. I noticed that you link to the old version of blue_hydra, and significant development has been done since the release in your readme. The new code can be found here : https://github.com/ZeroChaos-/blue_hydra

I have no idea how you are using it, but I recently added a file based json API for another project which you may like.  Just @ me if you need support.

ghostop14, 2019-11-07:
Thanks for the info ZeroChaos!  I'll grab the updated version and make sure all is well and update the link in the docs after I test it.

ghostop14, 2019-11-07:
Just tested and updated the README.  Thanks!

closed 2019-11-07 by ghostop14