ghostop14/sparrow-wifi #25: Add Channel Utilization and Station Count Columns

curtisklarsen, 2019-11-28:
Hello,

I use a different tool to scan WiFi networks on Linux, but I have found sparrow-wifi to be useful for sorting and visualizing the RSSI data by importing from CSV.  My tool gathers channel utilization and station count but sparrow-wifi has no columns for this.  Would you consider adding these columns, or a way to add custom columns?

Thanks.

ghostop14, 2019-12-02:
HI curtisklarsen, sure looks like it's in the data I pull.  I should have time over the weekend to add and test.  I'll repost when it's in.

ghostop14, 2019-12-03:
Just pushed updates including utilization and station count.  If the AP isn't providing the info, you'll have a -1 in each field as an indicator that the information is not available.

closed 2019-12-03 by ghostop14