ghostop14/sparrow-wifi #27: PyQtChart error

catalinalb, 2020-03-31:
while installing the second command line I got this error:

  Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.6/tokenize.py", line 452, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-vvvpeu8r/PyQtChart/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-vvvpeu8r/PyQtChart/

Any ideas on how to pass over it?
Thank you

ghostop14, 2020-03-31:
Most of those I've found are related to outdated pip.  Not sure what the new version does differently.  Try "sudo pip3 install --upgrade pip".  Then rerun the pip3 installs and see if it goes through.

catalinalb, 2020-03-31:
Thank you so much for your valuable and fast answer! 
I was able to start the program. The only thing now is that it shows an error for the hacrf_sweep in the terminal. Not sure why since qspectrumanalyzer is working great with my hackrf. I also have gr-fosphor and gr-gsm installed and working fine....

Thanks,
Catalin

closed 2020-03-31 by catalinalb