ghostop14/sparrow-wifi #28: File "sparrow-wifi.py", line 37, in <module>

gunner4361, 2020-04-28:
python3 sparrow-wifi.py 
Traceback (most recent call last):
  File "sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ModuleNotFoundError: No module named 'PyQt5.QtChart'

sudo pip3 install QScintilla PyQtChart gps3 dronekit manuf python-dateutil numpy matplotlib
Requirement already satisfied: QScintilla in /usr/local/lib/python3.8/dist-packages (2.11.4)
Requirement already satisfied: PyQtChart in /usr/local/lib/python3.8/dist-packages (5.14.0)

It says PyQtChart is there.

stryngs, 2021-05-20:
Sounds like a python environment issue to me.

nasa29, 2023-02-07:
I am not able to run sparrow-wifi.py  its showing error message
File "/home/nasa/sparrow-wifi/sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ImportError: /home/nasa/.local/lib/python3.10/site-packages/PyQt5/Qt5/lib/libQt5Charts.so.5: undefined symbol: _ZdlPvm, version Qt_5

akwala, 2024-07-05:
Installed in a virtualenv, after updating _pip_. Same result before and after `pip3 install pyqtchart==5.13.1`. 

Operating System: Kubuntu 22.04
KDE Plasma Version: 5.24.7
KDE Frameworks Version: 5.98.0
Qt Version: 5.15.3

```
$ sudo ./sparrow-wifi.py 
Traceback (most recent call last):
  File "/var/ak/dev/workspace/github/sparrow-wifi/./sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ModuleNotFoundError: No module named 'PyQt5.QtChart'
```

This is what's installed:
```
$ pip list
Package            Version
------------------ -----------
certifi            2024.7.4
charset-normalizer 3.3.2
contourpy          1.2.1
cycler             0.12.1
fonttools          4.53.0
gps3               0.33.3
idna               3.7
kiwisolver         1.4.5
matplotlib         3.9.1
numpy              2.0.0
packaging          24.1
pillow             10.4.0
pip                24.1.1
pyparsing          3.1.2
PyQt5              5.15.10
PyQt5-Qt5          5.15.14
PyQt5-sip          12.13.0
PyQtChart          5.13.1
PyQtChart-Qt5      5.15.14
python-dateutil    2.9.0.post0
requests           2.32.3
setuptools         70.2.0
six                1.16.0
urllib3            2.2.2
```