ghostop14/sparrow-wifi #29: click "Scan": no wireless adapters found

XiangyuTang, 2020-05-13:
Hi ghostop14, thank you very much for your great contribution first!
The problem I met cover the title.
When I run "python3 sparrow-wifi.py", I got the information:
"QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root' "
then the interface is activated, but can't scan any AP.
the program runs at Ubuntu 14.04, and kernel version: low-latency 3.19.0-61

Besides, do you want to add the FTM result of each AP based on 802.11mc? 
If I want to do some incremental development on your project, can you offer any good advice?

:)
thanks a lot!

ghostop14, 2020-05-14:
Hi, I've seen similar results when I accidentally leave my wifi in airplane mode or turned off.  It can look like it's scanning but no results.  A quick test would be to run something like "sudo iw dev <wlan adapter name> scan" and verify you get results and not an error message of any kind.

In terms of FTM, I do think that would be a nice addition.  If you happen to be able to run that scan command and send me the output that includes the FTM field, I'm happy to add it.  Wouldn't take too long.

XiangyuTang, 2020-05-14:
I've tried "iw dev wlan1 scan" command ,the results showed that everything was working fine. But still can't scan any AP on the interface. It's interesting and strange.

For FTM, I've already deployed my equipment and passed the ranging test, which refered to the website: http://www.winlab.rutgers.edu/~gruteser/projects/ftm/index.htm, you can have a try:)
It seems better if I can visualize the FTM result, so that's why I find you~

ghostop14, 2020-05-14:
Just doublechecking, is sparrow running with sudo?  I've tested on Ubuntu 16, 18, and straight debian, but never on 14.  Always possible something in the iw dev scan output is formatted differently.

XiangyuTang, 2020-05-15:
I actually ran with sudo, and I tested with different version of iw , there still comes to the same problem. Maybe Ubuntu 14 can't work well.

ghostop14, 2020-05-15:
Quite possible.  I haven't tested it on anything older than Ubuntu 16 and I know every once in a while the command-line tools can vary their output format with upgrades.  Quite possible it looks different.  There's regex's in the wirelessengine.py file that you can see I use for the attribute extraction.  Can always check them by hand to see if the regex just doesn't match the iw output on Ubuntu 14.

XiangyuTang, 2020-05-15:
I agree with you. Later I'll study the wirelessengine.py file and try to make some adaptation changes. If I make any progress I'll let you konw :)

Thanks~

closed 2020-05-15 by XiangyuTang