ghostop14/sparrow-wifi #30: (Request) OpenWRT agent similar to kismet-remote-cap

alphafox02, 2020-05-17:
More of a suggestion than anything else. I’d be nice if there was any way to build a remote agent for openwrt similar to what’s available for the newer kismet. Essentially kismet remote can push the WiFi and Bluetooth info to a kismet server. It’s a small footprint piece that runs on openwrt. I realize this would probably take a lot of changes on your end since most of what you’re doing seems to be python.. but maybe there’s an agent you could come up with after taking a look at what kismet does. The advantage you have is that your agent currently supports sending gps to the server, kismet remote does not (except static configured gps).

ghostop14, 2020-05-18:
Hi alphafox02, I haven't messed with the kismet agents, but the sparrowwifiagent.py is quite capable.  I actually just pushed a supplementary agent (sparrow-elastic.py) over the weekend that allows the wifi and bluetooth data to be uploaded to an Elasticsearch server / ELK stack in standard ECS-compliant format.  That opens up a world of opportunities for data trend analysis, visualization, and event WIDS / bluetooth IDS when combined with ElastAlert.  The data can also be combined in ELK's SIEM modules since it's ECS-compliant.

alphafox02, 2020-05-18:
There’s a few videos I put together on YouTube recently showing off some of sparrow-WiFi’s features so hopefully I’m doing it justice. Just have a look on YouTube for “cemaxecuter”. 

I was just thinking if there’d be a way to make an agent version capable of running on openwrt, that’d also be great. That may require something completely different coding wise. 

I’ve seen the recent updates and do what to try it with amazon. Correct me if I’m wrong, it seems that only the access points (WiFi) are identified and not clients connected? I may have missed something. 

At any rate, if you get a chance check out there one video I put up of kismet remote running on openwrt. It actually pushes the data out to a server instead of the other way around with the server reaching out to the agent. 

Cheers!

alphafox02, 2020-05-18:
I doubt openwrt has every dependency needed to run the .py file, but I’ll take a look. That’d be awesome if it did.

ghostop14, 2020-05-18:
alpha, the video sounds great.  I'll definitely check it out.  For the WiFi, it is just the AP or ad-hoc side.  I do have a plugin for sparrow I haven't released just due to problems that can arise that leverages aircrack-ng under the covers to get the connected clients as well.

alphafox02, 2020-05-20:
I’ll keep a look out for possible aircrack integration. That’d be great too.

alphafox02, 2020-12-10:
I forgot I posted this too. Was the aircrack-Ng issues ever resolved?

ghostop14, 2020-12-10:
I do have "plugins" that leverage aircrack-ng under the covers.  But I haven't open-sourced them.  Maybe I'll work on it a bit over the holiday.

alphafox02, 2020-12-10:
That’d be awesome! Hope you enjoy the holidays.

ghostop14, 2020-12-20:
@alphafox02 I just released the "Falcon" plugins to integrate aircrack-ng.  Check out the latest push and the updated README.  :)

alphafox02, 2020-12-20:
I can’t wait to check this out. Thankfully I brought my laptop and radios with me!

alphafox02, 2020-12-20:
It was literally drop in on DragonOS. Falcon menu is running and interface had no issue going to monitor mode. There was some pop up about John something wpa being better but not found.. I’ll see what that was talking about and add it. I see clients as expected. Awesome add on!

alphafox02, 2020-12-20:
Only thing I notice is that doing a reclick dearth on my AP seems to freeze the GUI, yet I can see the files in tmp are bro written to and ps auxx shows airodump ng and air crack running. I eventually kill the processes to get close out the GUI and restart.

alphafox02, 2020-12-21:
Building John and placing the relevant file in the /usr/bin/ folder keeps the GUI from freezing when attempting to capture keys - although I’ve not seen it capture one yet. The continuous deauth works great on my AP though.

ghostop14, 2020-12-21:
Awesome @alphafox02.  I'll update the README about john too.  I scrolled through the code to see if I captured all the external referenced tools, but guess I missed that one.  (I wrote this code about 2 years ago, I just never released it)

alphafox02, 2020-12-21:
It’s working great. Unfortunately I do not have all my equipment with me to test gps and what not. I’m wondering, seems to indicate so when looking at the new files, but if this is running on the pi or whatever, can you connect to the remote agent and also have the remote agent run the falcon tools? If so, will the clients also have gps info associated with them? That’d be awesome, not to mention if you can make the remote agent do other active pen testing things..

ghostop14, 2020-12-21:
Yes, with the agent, everything can run remote.  Even the Falcon stuff.  And it'll use the gps on the remote.  Basically all the functionality happens on the agent and the UI would just be for display.  All the prerequisites would apply on the remote system running the agent.

closed 2021-07-31 by alphafox02