ghostop14/sparrow-wifi #32: Incorrect Channel Widths

curtisklarsen, 2020-06-06:
I have an 802.11ac access point on channel 165 that is 20MHz wide.  It has multiple SSIDs, sparrow-wifi detects some of the SSID's correctly as 20MHz, and others incorrectly as 40MHz wide.

stryngs, 2021-05-20:
Wifi and width can be difficult, especially as more protocols are accepted into the mainstream.  AC being one of them.

ghostop14, 2021-05-21:
curtisklarsen if you can post the iw output, I can take a look through it for the width info.  First thought is I wonder if it's splitting between two different channels (2.4 GHz accounts for that).  Or it could have to do with the output and the default if the OS doesn't report anything.  Also, what OS are you on?  Also if you can indicate what SSID you know it's reporting incorrectly, that'll help too.  (I can debug with the iw output)

drws, 2022-11-02:
There hasn't been any activity from @curtisklarsen for two years. There's a good chance that the issue was resolved inbetween or that the original issue can be explained by [inner workings of Wi-Fi](https://github.com/ghostop14/sparrow-wifi/issues/33#issuecomment-1299718301). Also, it might simply be a bug in the AP.

Since the info is old and incomplete I suggest this is closed.