ghostop14/sparrow-wifi #34: no qt platform plugin could be initalized

ImeEon, 2020-07-29:
I get this error when running sparrow-wifi.py. I finished the installation in readme.md. 
   
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized.
Available platform plugins are: eglfs, linuxfb ...
Aborted ( core dumped )

Biedubbeljoe, 2020-09-18:
I have the same issue, using WSL2 on the Windows 10 subsystem (win-kex):
~/sparrow-wifi$ sudo python3 sparrow-wifi.py 
[sudo] password for xxxxxxx: 
No protocol specified
qt.qpa.xcb: could not connect to display :1.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
Aborted
-------
anyone who has solved this issue?

ghostop14, 2020-09-18:
I've never tried running it on WSL, so could be some WSL-specific things going on.  Could try an Ubuntu or kali VM instead.

Biedubbeljoe, 2020-09-19:
Totally agree with you. But wouldn't it be nice if someone stumbled into this issue, has the answer and is willing to share this with us?

ghostop14, 2020-09-20:
Absolutely.  If anyone has it running in WSL post here.  I may try too over the next week if I can get WSL up on one of my systems.

mefromthepast, 2021-08-19:
Same issue here with Archlinux / sway (wayland).

```
sparrow-wifi                  
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

/usr/bin/sparrow-wifi: line 3: 133103 Aborted                 (core dumped) pkexec /opt/sparrow-wifi/sparrow-wifi.py
```

ghostop14, 2021-08-19:
Looks like you may be connecting over SSH?  The "could not connect to display :0" means it couldn't access the GUI context.  You'll also need to run with sudo.  I'd suggest trying it local first, then if you move over to SSH use -X to allow GUI apps to load over SSH.  Another option would be to run the agent on the remote system then just connect the UI to the agent for remote operations.

mefromthepast, 2021-08-19:
Thanks for the reply, but I'm running locally with `polkit`. Output for `sudo sparrow-wifi` is exactly the same.

ghostop14, 2021-08-19:
Try sudo -i first, then run it from there.  See if it works any better (that display :0 looks like the part to get past).

mefromthepast, 2021-08-20:
The solution has been executing `xhost +si:localuser:root`. (Sorry for hijacking this thread, I thought that it might have been a more closely related issue that not solely depends on wayland.)

mb720, 2022-03-23:
Executing `xhost +si:localuser:root` also fixed the issue for me on Sway (Wayland), thanks @mefromthepast 👍

I think this issue can be closed as it doesn't pertain to sparrow-wifi per se: For example, [LinSSID has the same issue and solution]( https://bugs.launchpad.net/linssid/+bug/1966079).

ghostop14, 2022-03-23:
Thanks folks!  I'll close it out.

closed 2022-03-23 by ghostop14