ghostop14/sparrow-wifi #40: ImportError Qt_5.15 not found

Biedubbeljoe, 2020-08-17:
Hi,
I tried to install sparrow-wifi on two different laptops following the install instructions. On both machines using the latest Kali and fully updated I get the same error as I run the python script.
kali:~/sparrow-wifi$ sudo ./sparrow-wifi.py
Traceback (most recent call last):
  File "./sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ImportError: /lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.15' not found (required by /usr/local/lib/python3.8/dist-packages/PyQt5/QtChart.abi3.so)

The libQt5Core.so.5 is present: kali:/lib/x86_64-linux-gnu$ ls libQt5Core.so.5
libQt5Core.so.5

How can I fix this issue?

ghostop14, 2020-08-17:
Hi biedubbeljoe,  Qtchart does seem to be notoriously stubborn these days.  Try some of these and see if it fixes it for you...

- The latest version of PyQtChart has been giving folks issues if you don't "sudo pip3 install --upgrade pip" first.  If you get an error installing any of the modules, upgrade pip and try it again.  They must be using something in the newest pip that breaks older versions.
- On Ubuntu 20.04+ do not pip3 install pyqtchart, rather, use 'sudo apt-get install python3-pyqt5.qtchart'
- If after installing below you still get some qtchart errors, some folks have had luck with pip3 install pyqtchart==5.13.1

algisju, 2020-08-20:
Same problem; any way round yet?
kali and all updates/upgrades installed: tried the given solution but to no avail...

Biedubbeljoe, 2020-08-20:
I took a fresh Kali install (latest), used the two install lines from the git added sudo apt python3-pyqt5.qtchartand it worked. All previous attempts failed.

ghostop14, 2020-08-20:
I'll have some time next week to look at it and sort it out.  I'll post a fix as soon as I have one.

algisju, 2020-08-20:
Thanks guys; I haven't tried reinstalling a latest kali; but I feel a bit silly to boot a vm only for a program. Cheers; god job!

ghostop14, 2020-08-20:
Can always try a docker.  Might be less resource-intensive.  I prefer VMs too but dockers aren't too bad for quick tests.  I'll figure out what it is and get a solution posted.

alphafox02, 2020-08-24:
If you’re going to try a VM, then try DragonOS Focal. 20.04 based and sparrow is already installed and working as shown here 

https://youtu.be/X73ZbZbfrRs

Biedubbeljoe, 2020-08-25:
Will have a look, thanks for that.

https://youtu.be/X73ZbZbfrRs

ghostop14, 2020-08-30:
Hi folks, so I grabbed a fresh Kali 2020.3 VM from offensive security and here's updated prerequisite install procedures:
sudo apt-get install python3-pip gpsd gpsd-clients python3-tk python3-setuptools python3-pyqt5.qtchart
sudo pip3 install QScintilla gps3 dronekit manuf python-dateutil numpy matplotlib

I pushed a readme update and a sparrow-wifi.py update to get rid of some deprecation warnings.

Let me know if that does the trick for you.  Seems like on newer OS's (Ubuntu 20+ and Kali 2020.3) you need to apt install python3-pyqt5-qtchart rather than doing it via pip.

Biedubbeljoe, 2020-08-31:
Thanks Mate, that's working for me!

algisju, 2020-08-31:
That does it! Thank You!