ghostop14/sparrow-wifi #45: Is it possible to use this on MacOS?

Kunkles, 2020-10-09:
Is there a way to use this with OSX? I've got a HackRf and I've been looking for a decent tool like this

ghostop14, 2020-10-13:
Hi Kunkles, I'm not a mac user so I don't know if anyone's gotten it to work on a mac.  I don't know that there's anything in the python code that wouldn't run on a mac.  The part I wouldn't know about would be some of the shelled out commands to collect data like the output formats from iwconfig and iw.  If they differ than other debian/Ubuntu versions, the regex's could fail.  Probably the easiest would be to just run an Ubuntu docker on the mac and run it in there.

geoffreysmith, 2021-08-08:
This is not possible without significant modification. This makes use of iw set of commands (as far as I know they're obsolete in Linux, but I could be wrong). I spent some time trying to get it to work with OS X and gave up.