ghostop14/sparrow-wifi #47: BlueTooth scann error

killerdevildog11, 2020-11-20:
python3 give this error

/usr/lib/python3.8/subprocess.py:844: RuntimeWarning: line buffering (buffering=1) isn't supported in binary mode, the default buffer size will be used
  self.stdout = io.open(c2pread, 'rb', bufsize)

when trying to scan bluetooth

ghostop14, 2020-12-10:
Hi killerdevildog11, finally upgraded my dev system to a platform with python3.8 so I'll take a look at it.  Reading the Google search results on it, I think it can be safely ignored as it's just a warning confirming behavior the call has always had.  But I'll look and see if it's causing any issues.  Thanks!

ghostop14, 2020-12-10:
Fixed.  Just pushed the fix.  Test it out and see if that takes care of it for you as well, but I'll close this out.

closed 2020-12-10 by ghostop14