ghostop14/sparrow-wifi #49: installation error

beesuns, 2020-12-28:
Kali Linux version: `Linux kali 5.9.0-kali5-amd64 #1 SMP Debian 5.9.15-1kali1 (2020-12-18) x86_64 GNU/Linux` 
pip version: `pip 20.3.3 from /home/kali/.local/lib/python3.9/site-packages/pip (python 3.9)`
qtchart installed: `python3-pyqt5.qtchart is already the newest version (5.15.2+dfsg-1).`

But when I run `sudo pip3 install QScintilla gps3 dronekit manuf python-dateutil numpy matplotlib`, it gave error message.
```
sudo pip3 install QScintilla gps3 dronekit manuf python-dateutil numpy matplotlib
Requirement already satisfied: python-dateutil in /usr/local/lib/python3.9/site-packages (2.8.1)
Collecting dronekit
  Using cached dronekit-2.9.2-py3-none-any.whl (56 kB)
Collecting monotonic>=1.3
  Using cached monotonic-1.5-py2.py3-none-any.whl (5.3 kB)
Collecting pymavlink>=2.2.20
  Using cached pymavlink-2.4.14.tar.gz (4.1 MB)
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-c4b35r3n/pymavlink_c4c59a4282224373856aa74c091c960f/setup.py'"'"'; __file__='"'"'/tmp/pip-install-c4b35r3n/pymavlink_c4c59a4282224373856aa74c091c960f/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-52ukk2lr          
         cwd: /tmp/pip-install-c4b35r3n/pymavlink_c4c59a4282224373856aa74c091c960f/                                                                                   
    Complete output (44 lines):                                                                                                                                       
    WARNING: The wheel package is not available.                                                                                                                      
      ERROR: Command errored out with exit status 1:                                                                                                                  
       command: /usr/local/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-a40v92ub/future_03194d7bcb2d4a139c9900febcd1ae20/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-a40v92ub/future_03194d7bcb2d4a139c9900febcd1ae20/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-mvb26vmx                           
           cwd: /tmp/pip-wheel-a40v92ub/future_03194d7bcb2d4a139c9900febcd1ae20/                                                                                      
      Complete output (6 lines):                                                                                                                                      
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]                                                                                           
         or: setup.py --help [cmd1 cmd2 ...]                                                                                                                          
         or: setup.py --help-commands                                                                                                                                 
         or: setup.py cmd --help                                                                                                                                      
                                                                                                                                                                      
      error: invalid command 'bdist_wheel'                                                                                                                            
      ----------------------------------------                                                                                                                        
      ERROR: Failed building wheel for future                                                                                                                         
    ERROR: Failed to build one or more wheels                                                                                                                         
    Traceback (most recent call last):                                                                                                                                
      File "/usr/local/lib/python3.9/site-packages/setuptools/installer.py", line 128, in fetch_build_egg                                                             
        subprocess.check_call(cmd)                                                                                                                                    
      File "/usr/local/lib/python3.9/subprocess.py", line 373, in check_call                                                                                          
        raise CalledProcessError(retcode, cmd)                                                                                                                        
    subprocess.CalledProcessError: Command '['/usr/local/bin/python3.9', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpuco1m1xx', '--quiet', 'future']' returned non-zero exit status 1.                                                                                                                 
                                                                                                                                                                      
    The above exception was the direct cause of the following exception:                                                                                              
                                                                                                                                                                      
    Traceback (most recent call last):                                                                                                                                
      File "<string>", line 1, in <module>                                                                                                                            
      File "/tmp/pip-install-c4b35r3n/pymavlink_c4c59a4282224373856aa74c091c960f/setup.py", line 109, in <module>                                                     
        setup (name = 'pymavlink',                                                                                                                                    
      File "/usr/local/lib/python3.9/site-packages/setuptools/__init__.py", line 164, in setup                                                                        
        _install_setup_requires(attrs)                                                                                                                                
      File "/usr/local/lib/python3.9/site-packages/setuptools/__init__.py", line 159, in _install_setup_requires                                                      
        dist.fetch_build_eggs(dist.setup_requires)                                                                                                                    
      File "/usr/local/lib/python3.9/site-packages/setuptools/dist.py", line 699, in fetch_build_eggs                                                                 
        resolved_dists = pkg_resources.working_set.resolve(                                                                                                           
      File "/usr/local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 779, in resolve                                                                   
        dist = best[req.key] = env.best_match(                                                                                                                        
      File "/usr/local/lib/python3.9/site-packages/pkg_resources/__init__.py", line 1064, in best_match                                                               
        return self.obtain(req, installer)                                                                                                                            
[... cut ...]
    distutils.errors.DistutilsError: Command '['/usr/local/bin/python3.9', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpuco1m1xx', '--quiet', 'future']' returned non-zero exit status 1.                                                                                                               
    ----------------------------------------                                                                                                                          
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
Please help me to fix this. Thanks!

ghostop14, 2020-12-28:
If you're not doing anything with drones and Mavlink, you can leave mavlink out.

beesuns, 2020-12-29:
@ghostop14 Thanks for your reply. I am not sure if this was related to above mentioned problem.
I got this error message when I tried to run sparrow-wifi:
```
kali@kali:~/Downloads/sparrow-wifi$ sudo ./sparrow-wifi.py 
[sudo] password for kali: 
Traceback (most recent call last):
  File "/home/kali/Downloads/sparrow-wifi/./sparrow-wifi.py", line 28, in <module>
    from dateutil import parser
ModuleNotFoundError: No module named 'dateutil'
```
Tried to install dateutil:
```
kali@kali:~/Downloads/sparrow-wifi$ sudo pip3 install python-dateutil
Requirement already satisfied: python-dateutil in /usr/local/lib/python3.9/site-packages (2.8.1)
Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.9/site-packages (from python-dateutil) (1.15.0)
```
Run again:
```
kali@kali:~/Downloads/sparrow-wifi$ sudo ./sparrow-wifi.py 
Traceback (most recent call last):
  File "/home/kali/Downloads/sparrow-wifi/./sparrow-wifi.py", line 28, in <module>
    from dateutil import parser
ModuleNotFoundError: No module named 'dateutil'
```

ghostop14, 2020-12-29:
beesuns you were on the right track.  Is it possible /usr/local/lib/python3.9 isn't in your PYTHONPATH?  Looks like it's already installed there.  Also looks like you have a python reference above to /home/kali/.local/lib/python3.9.  Sounds like your python path may be the problem.