ghostop14/sparrow-wifi #52: Bladerf support (request)

alphafox02, 2021-01-15:
Possibly using the monitor mode interface created for sparrow WiFi might would be really awesome. You’d have like three SDRs going at one time with the Uber tooth hackrf and bladerf.

https://www.nuand.com/bladerf-wiphy/

There is some recent submits to kismet that allow it to work. I’m waiting to get hands on an xa9. Excited to test.

ghostop14, 2021-01-15:
That is an interesting project.  One shortcoming I've run into in years past are WIFI signals that use bandwidths supported by the spec but not by most chipsets.  I actually have a source here (a drone) that I think meets that criteria.  It doesn't show up on normal wifi chipsets, but I know it's a little-used wifi bandwidth.  But I don't have a bladerf to test with though either.

alphafox02, 2021-01-27:
So I thought I’d get lucky and after setting up bladerf-wiphy but prior to running hostapd to create an open access point.. I fired up sparrow WiFi and could see wlan1 selectable (the bladerf) however I’m sure there’s way more to this then I understand. It didn’t scan haha.. but the kismet guys have built a way to directly get the scans and stuff from the bladerf fpga. That’s next on my list to test. Here’s a video of just the bladerf and the open hotspot parts 
DragonOS Focal BladeRF-wiphy w/ Open Wi-Fi AP and Splash page https://youtu.be/1hUK2iXQtCI

ghostop14, 2021-01-27:
Interesting.  From a command-line you can try this to get an idea if a basic scan would work:
```iw dev wlan1 scan```  
Changing the wlan as needed.  That's what gets triggered and parsed behind the scenes.

alphafox02, 2021-01-28:
sudo iw dev wlan1 scan doesn’t work but wlan1 info works. Just shows the type, channel, tx power and some other info. Super cool, but I’m sure there’s lots more to do.

alphafox02, 2021-01-28:
What’s cool though is I used airodump-ng with a set channel after putting it in monitor mode and got some stuff on channel 1. I’ll go try the falcon tools and see what happens

alphafox02, 2021-01-28:
Ha, it seems to have worked. I got one thing back on channel 3. All fields are filled in except vendor and SSID is unknown. I’ll hook up an rx amp.

ghostop14, 2021-01-28:
Nice!

alphafox02, 2021-02-13:
Turns out the bladerf can be used just as any other monitor mode interface so long as you load bladerf wiphy and complete the steps required to have the device setup and running as a monitor mode interface. Then within the Falcon tools you can use the interface, which I was surprised to find this interface captured a handshake. Haven’t confirmed it can deauth but I suspect it’ll work.

closed 2021-02-13 by alphafox02

ghostop14, 2021-02-16:
Awesome!