ghostop14/sparrow-wifi #54: (Request) czml / cesium app

alphafox02, 2021-02-14:
An idea I thought I’d pitch at you as well. Similar to Df-aggregator’s use of cesium, I wondered if you’d be interested (I can’t code yet) in making a way for the remote agent part of sparrow WiFi to be added onto cesium while also being controlled by the server, etc. 

For example, DF-Aggregator is an app I guess you’d say that overlays/interacts with cesium either online or offline. I’ve built in both sparrow WiFi and an offline cesium setup to DragonOS. Right now DFA reaches out to the IPs of devices running KerberosSDr software and pulls in the xml formatted info and displays it on cesium. In this case it’s in the form of either stationary or mobile sensors with lines of bearing drawn on the map. 

I was picturing another sparrow WiFi python script that could do something similar with the remote agents locations and all the wireless devices being entered into a database similar to DFA and then shown on the map. This could also allow multiple remote agents to feed one database or even maybe be controlled via the overlay/app created for cesium.

https://github.com/ckoval7/df-aggregator

ghostop14, 2021-02-23:
I wrote the agent to be very flexible.  It's just HTTP requests with JSON responses so it can definitely be bridged into other solutions.  There's also a sparrow-elastic.py script included now that can write data directly into Elasticsearch.  Then you can use Kibana or Grafana to create dashboards based on it.  I have an agent running 24x7 here that feeds Elastic with some wifi and bluetooth dashboards that I can access any time.  You could feed multiple agents in, etc. for a distributed view.

alphafox02, 2021-02-28:
Definitely need to try the elastic search piece. One of these days I hope to be able to contribute more, once I actually learn python c etc!

alphafox02, 2021-03-04:
I’m going to start going down this rabbit hole. So do you or can you run elasticsearch piece on your own server? I was looking at the setup and figure I can host it myself and try the sparrow scripts.

alphafox02, 2021-03-17:
@ghostop14 i was able to setup elasticsearch and Kibana as you mentioned with both the WiFi and Bluetooth working and being indexed. This is really awesome. I haven’t added gps yet to check out the map portion. I was wondering, don’t think so after I looked over the agent, but can/is the elasticsearch agent capable of taking in the Falcon advanced scan info for WiFi?

alphafox02, 2021-03-17:
I see this mentioned but thought maybe if I used the GUI to change or add the mon interface in the Falcon tools section that I’d mess up how I told the sparrow elasticsearch agent to start. 

“ See the Falcon/aircrack section for details.

Elasticsearch Database Integration: An additional agent has been added that can feed wireless and bluetooth network scan results into an Elasticsearch database. See the Elasticsearch section for details.”

closed 2021-07-31 by alphafox02