ghostop14/sparrow-wifi #56: Elasticsearch agent without Bluetooth

alphafox02, 2021-03-17:
In testing, it appears not possible to run the sparrow elasticsearch agent when there’s no Bluetooth present either by design or by mistake. Since it mentions using a Bluetooth index it would appear the intent is meant to run it with or without Bluetooth for WiFi only operation. 

However, when starting without a btindex it complains one is missing, but then when one is specified and no Bluetooth is present it then tells you it can run because there’s no Bluetooth capability.

ghostop14, 2021-03-18:
Hmmm, I'll look into that.  The intent would be to be able to run with either/or or both.

ghostop14, 2021-03-18:
I just pushed an update to not create the bt index if no command-line bt index is specified.  But I do check for bluetoothEnabled.  Try the latest and see if it still does it.

alphafox02, 2021-03-19:
Works perfect with or without Bluetooth now . If I leave off the bt index it starts and runs with just the WiFi interface.

closed 2021-03-19 by alphafox02