ghostop14/sparrow-wifi #57: Problem with bluetooth output

jwlowmiller, 2021-03-23:
I can't seem to get any devices to show up in Sparrow for bluetooth. 
If I run bluetoothctl  with scan on I get a stream of devices found. 

<pre><font color="#739FCF">[bluetooth]</font># scan on
Discovery started
[<font color="#FCE94F">CHG</font>] Controller AC:67:5D:CA:AE:3A Discovering: yes
[<font color="#8AE234">NEW</font>] Device A4:C1:38:32:50:FB ihoment_H6142_50FB
[<font color="#8AE234">NEW</font>] Device 5C:ED:D2:AC:4C:FB 5C-ED-D2-AC-4C-FB
</pre>

But with sparrow I cannot.

<pre><span style="background-color:#3465A4"><font color="#E4E4E4"> /</font></span><span style="background-color:#3465A4"><font color="#EEEEEE"><b>opt</b></font></span><span style="background-color:#3465A4"><font color="#E4E4E4">/</font></span><span style="background-color:#3465A4"><font color="#EEEEEE"><b>sparrow-wifi</b></font></span><span style="background-color:#3465A4"><font color="#E4E4E4"> </font></span><span style="background-color:#4E9A06"><font color="#3465A4"></font></span><span style="background-color:#4E9A06"><font color="#000000"> master </font></span><font color="#4E9A06"></font> <font color="#4E9A06"><u style="text-decoration-style:single">sudo</u></font> <font color="#4E9A06">hcitool</font> lescan                                                      <font color="#CC0000"></font><span style="background-color:#CC0000"><font color="#C4A000"> 1 ✘ </font></span><span style="background-color:#CC0000"><font color="#D3D7CF"></font></span><span style="background-color:#D3D7CF"><font color="#000000"> 19:18:34 </font></span>
Set scan parameters failed: Input/output error
</pre>

alphafox02, 2022-07-21:
I’m having what seems to be a similar issue. I can see there’s some interaction with hci when running the Bluetooth scan (with standard adapter) as dmesg shows some activity. How’d you locate the failing output you show above? Did you ever resolve the issue..

What’s interesting is that on my raspberry pi running 20.04 as well, nearly all the same software that I can tell except it’l has a 5.4 kernel instead of a 5.15 like my laptop I do in fact get normal Bluetooth scan output with sparrow WiFi on it. 

Also on the laptop all other tools can generate Bluetooth scan results.

alphafox02, 2022-07-21:
@ghostop14 i have two laptops, both have internal Bluetooth. I now confirmed with the same exact image booted and ran live on each laptop I get the following results.

1. Runs sparrow WiFi and normal Bluetooth scan yields results, this is with 20.04 and the latest hwe 5.15 kernel.

2. Second laptop, same software/image Sparrow WiFi does not yield normal Bluetooth scan results, yet hci and blue hydra both generate scan results. 

I would love to provide more info, only thing I see I think different on second laptop is that each time I try to do a scan it shows this in dmesg 

Bluetooth: hci0: MSFT filter_enable is already on. 

I don’t know where else to debug this but it seems like maybe a way the Bluetooth scan is being conducted it is not liked in some cases.

alphafox02, 2022-07-21:
It just occurred to me. If this even makes sense, maybe the Bluetooth in the one laptop isn’t standard and thus the normal scan sparrow is doing (without ubertooth) is not compatible?

alphafox02, 2022-07-21:
I was wrong, I to get a  failure as mentioned above with sudo hcitool lescan, however blue hydra works. It appears after further research it may be associated to the kernel and people are saying bluetoothctl should maybe be used instead of hci.

alphafox02, 2022-07-21:
“bluetoothctl scan on” ran in terminal appears to be a replacement for hci for btle 5.x devices and works great where hci fails. 

Maybe it could be an optional means of scanning. However maybe this would require some extensive changes to the scanning mechanism.