ghostop14/sparrow-wifi #68: (Request) Aircrack NG (falcon tools) 6E channel/freq scans

alphafox02, 2022-04-18:
Can the ability to scan 6E be scripted in? I haven’t looked closely yet, but according to this post it should be possible for aircrack to be set to frequency in MHz, maybe that would require a manual table of info to scan through? Or maybe aircrack will have a 6e band added soon.

https://github.com/aircrack-ng/aircrack-ng/issues/2205

Also, 

Would it be possible for the sparrow-WiFi agent to start with the falcon mode of scanning enabled? Meaning without having to connect with a GUI to activate it?

ghostop14, 2022-04-27:
I think if aircrack scans it, it should come out.  I don't have any 6E here yet to test it out with, but may get some soon.

For the falcon start, there isn't a param or anything for it since there's a requirement for the monitor mode on the card.  You could just use curl or wget and trigger it in a launch script after you start the agent.  That's probably the easiest solution.

alphafox02, 2022-04-27:
I was thinking maybe a parm or like you mentioned, maybe something additional that would put the elasticsearch py agent automatically into monitor mode (if supported) would allow little to no interaction by the user to setup the agent in “falcon” mode.

alphafox02, 2022-05-29:
Some good news here on aircrack-ng and 6e channels. A new release includes the ability along with wpa3 etc. 

https://aircrack-ng.blogspot.com/2022/05/aircrack-ng-17.html?m=1

alphafox02, 2022-05-29:
Looks like you have to specify specific channels for now though “sudo airodump-ng wlan1mon -w 6E_test -C 6115”. Maybe an option could be added to the Falcon GUI to set specific channels on/off when other then band scanning is needed.

NorthHat, 2022-06-06:
Hello friends!

I used Wi-Fi adapter Intel AX210NGW in Ubuntu 22.04 (kernel: 5.15.0-30) and driver "backport-iwlwifi-dkms":

[spoiler]
[code]
linux@linux-MS-7B25:~$ sudo dmesg | grep -i iwlwifi
[    3.613872] Loading modules backported from iwlwifi
[    3.613872] iwlwifi-stack-public:master:9858:4c7cba27
[    3.669612] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[    3.686624] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-73.ucode failed with error -2
[    3.686638] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-72.ucode failed with error -2
[    3.690686] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[    3.690699] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
[    3.691035] iwlwifi 0000:01:00.0: loaded firmware version 71.058653f6.0 ty-a0-gf-a0-71.ucode op_mode iwlmvm
[    3.820928] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    3.970859] iwlwifi 0000:01:00.0: loaded PNVM version 05a8dfca
[    3.982159] iwlwifi 0000:01:00.0: Detected RF GF, rfid=0x10d000
[    4.050969] iwlwifi 0000:01:00.0: base HW address: f4:46:37:cb:56:b5
[    4.105650] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
[  490.546418] UBSAN: shift-out-of-bounds in /var/lib/dkms/backport-iwlwifi/9858/build/drivers/net/wireless/intel/iwlwifi/mvm/mac-ctxt.c:668:22

linux@linux-MS-7B25:~$ iw phy0 reg get
phy#0
country US: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5190 - 5210 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS
	(5210 - 5230 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5230 - 5250 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS
	(5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5490 - 5510 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5510 - 5530 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5530 - 5550 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5550 - 5570 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5570 - 5590 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5590 - 5610 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5610 - 5630 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5630 - 5650 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-160MHZ
	(5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-160MHZ
	(5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-160MHZ
	(5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-160MHZ
	(5815 - 5835 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(5945 - 7065 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, PASSIVE-SCAN
	(7065 - 7105 @ 40), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN
	(7105 - 7125 @ 20), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN
 
		Frequencies:
			* 2412 MHz [1] (22.0 dBm)
			* 2417 MHz [2] (22.0 dBm)
			* 2422 MHz [3] (22.0 dBm)
			* 2427 MHz [4] (22.0 dBm)
			* 2432 MHz [5] (22.0 dBm)
			* 2437 MHz [6] (22.0 dBm)
			* 2442 MHz [7] (22.0 dBm)
			* 2447 MHz [8] (22.0 dBm)
			* 2452 MHz [9] (22.0 dBm)
			* 2457 MHz [10] (22.0 dBm)
			* 2462 MHz [11] (22.0 dBm)
			* 2467 MHz [12] (22.0 dBm)
[... 117 lines cut ...]

[code]
sudo airodump-ng wlp1s0mon -c 1
[/code]

this is working normal,

but when i use the command:

[code]
sudo airodump-ng wlp1s0mon -C 2412
[/code]

result:

[code]
linux@linux-MS-7B25:~$ sudo airodump-ng wlp1s0mon -C 2412
Checking available frequencies, this could take few seconds.
Done.
No valid frequency given.
[/code]

this problem only then use driver "backport-iwlwifi-dkms" !

how fixed this problem ?

alphafox02, 2025-10-30:
Started to think about this again since on day 24.04, it’s capable of connecting to such channels with the correct hardware and both iw and aircrack both provide results. I feel like with small pin point changes it won’t be too difficult to add 6e support - and with additional work perhaps a 3rd graph can display results (minus fft SDR spectrum). 

I’d like to slowly work on this time permitting and keep it as close as possible to how sparrow WiFi handles 2.4 and 5Ghz currently.