ghostop14/sparrow-wifi #74: Traceback when BlueTooth Discovery Clicked

alphafox02, 2022-09-17:
Setup

22.04 aarch64 Ubuntu w/ all installed packages except *QScintilla*, which is just being a pain and not installing. I don't think however that's causing the issue I'm seeing, but then again I could be wrong.

Expectation

Click bluetooth discovery and be able to use the internal available Pi4 bluetooth. Verified hcitool lescan produces results.

What actually happens when the button is clicked as shown in the terminal

```
Traceback (most recent call last):
  File "/usr/src/sparrow-wifi/./sparrow-wifi.py", line 1455, in onBtBluetooth
    self.bluetoothWin = BluetoothDialog(self, self.bluetooth, self.remoteAgentUp, self.remoteAgentIP, self.remoteAgentPort, None)  # Need to set parent to None to allow it to not always be on top
  File "/usr/src/sparrow-wifi/sparrowdialogs.py", line 1187, in __init__
    self.resize(self.mainWidth, self.mainHeight)
TypeError: arguments did not match any overloaded call:
  resize(self, QSize): argument 1 has unexpected type 'float'
  resize(self, int, int): argument 1 has unexpected type 'float'
```

ghostop14, 2022-09-20:
Just pushed a fix.  Give that a shot.

alphafox02, 2022-09-21:
Thank you for taking a look. What I observe now is that the bluetooth discovery page does work, however, even after plugging in a GPS and getting green (actually both with and without GPS) the telemetry window will not open. It shows this in the terminal,

`/usr/src/sparrow-wifi/telemetry.py:92: UserWarning: FixedFormatter should only be used together with FixedLocator
  self.axes.set_yticklabels(['-20', '-40', '-60', '-80', '-100'], color=self.fontColor)
Traceback (most recent call last):
  File "/usr/src/sparrow-wifi/./sparrow-wifi.py", line 1886, in onShowTelemetry
    telemetryWindow = TelemetryDialog()
  File "/usr/src/sparrow-wifi/telemetry.py", line 155, in __init__
    self.createTable()
  File "/usr/src/sparrow-wifi/telemetry.py", line 184, in createTable
    self.locationTable.setGeometry(10, 10, self.geometry().width()/2-20, self.geometry().height()/2)
TypeError: arguments did not match any overloaded call:
  setGeometry(self, QRect): argument 1 has unexpected type 'int'
  setGeometry(self, int, int, int, int): argument 3 has unexpected type 'float'`

It also causes the GUI to "lock up" after clicking to obtain telemetry on a selected network.

ghostop14, 2022-09-21:
Looks like I missed one more.  Just pushed an update to telemetry.py.  Give that a shot. What's happening is the newer python versions are taking <int> / <int> and returning a float rather than an int.  So changing the division operator to a // tells it to treat it as integer math and return an int.  

Let me know if you run into any more.

alphafox02, 2022-09-21:
I think it's almost there, the telemetry window does open for WiFi, but not Bluetooth (I think I remember it doing both?).. but also the window that opens for the selected WiFi network is formatted funny and says this in the terminal. When highlighting a Bluetooth network and clicking telemetry, nothing happens at all in the terminal or elsewhere.

`/usr/src/sparrow-wifi/telemetry.py:92: UserWarning: FixedFormatter should only be used together with FixedLocator
  self.axes.set_yticklabels(['-20', '-40', '-60', '-80', '-100'], color=self.fontColor)
Traceback (most recent call last):
  File "/usr/src/sparrow-wifi/telemetry.py", line 225, in resizeEvent
    self.radar.setGeometry(self.geometry().width() - smallerDim - 10, 10, smallerDim, smallerDim)
TypeError: arguments did not match any overloaded call:
  setGeometry(self, QRect): argument 1 has unexpected type 'float'
  setGeometry(self, int, int, int, int): argument 1 has unexpected type 'float'`

[image]

alphafox02, 2022-09-21:
Messing around with changing 217/218 and maybe 231-236ish or so. Not sure if I need to be changing all the / to // without really looking close at what I'm doing.

ghostop14, 2022-09-21:
Just tested here, and bluetooth telemetry worked okay.  If you go into bluetooth and have a scan running, if you right-click on the item you want to track in telemetry and select it from that right-click menu it should come up.  If a scan isn't running it won't really change, but will update when a scan is running.

And that warning looks like something new from matplotlib.  I'll have to look into the correct way to address it.  But for now you can just ignore it.

alphafox02, 2022-09-21:
I was going about it wrong, forgot to right click. Instead I was clicking back on the main menu, but once I right clicked I found the next instance(s) of / on line 558 in telemetry.py. After making that change, all is working nicely so far, minus the matplotlib thing. I think it's safe to close this ticket.

closed 2022-09-21 by alphafox02