ghostop14/sparrow-wifi #79: Program crashes when Telemetry menu is accessed

drws, 2022-11-02:
I cannot use the _Telemetry_ menu in the latest Sparrow-WiFi, built [from AUR](https://aur.archlinux.org/packages/sparrow-wifi-git). Also just tried to rebuild, but it's the same. The program crashes with:

```
/opt/sparrow-wifi/telemetry.py:92: UserWarning: FixedFormatter should only be used together with FixedLocator
  self.axes.set_yticklabels(['-20', '-40', '-60', '-80', '-100'], color=self.fontColor)
Traceback (most recent call last):
  File "/opt/sparrow-wifi/telemetry.py", line 225, in resizeEvent
    self.radar.setGeometry(self.geometry().width() - smallerDim - 10, 10, smallerDim, smallerDim)
TypeError: arguments did not match any overloaded call:
  setGeometry(self, QRect): argument 1 has unexpected type 'float'
  setGeometry(self, int, int, int, int): argument 1 has unexpected type 'float'
/usr/sbin/sparrow-wifi: line 3: 140330 Aborted                 (core dumped) pkexec /opt/sparrow-wifi/sparrow-wifi.py
```

ghostop14, 2022-11-06:
I pushed an update a couple weeks ago to address this.  Try pulling/running from master.

closed 2022-11-06 by ghostop14

roseninc, 2022-11-16:
I have a similar issue when trying to access the telemetry for a specific target network, using the latest version:

/home/kali/sparrow-wifi/telemetry.py:92: UserWarning: FixedFormatter should only be used together with FixedLocator
  self.axes.set_yticklabels(['-20', '-40', '-60', '-80', '-100'], color=self.fontColor)
Traceback (most recent call last):
  File "/home/kali/sparrow-wifi/telemetry.py", line 225, in resizeEvent
    self.radar.setGeometry(self.geometry().width() - smallerDim - 10, 10, smallerDim, smallerDim)
TypeError: arguments did not match any overloaded call:
  setGeometry(self, QRect): argument 1 has unexpected type 'float'
  setGeometry(self, int, int, int, int): argument 1 has unexpected type 'float'
zsh: IOT instruction  ./sparrow-wifi.py