ghostop14/sparrow-wifi #80: Replace depracatred iwconfig to for example to nmcli.

chiddekel, 2022-11-05:
Replace deprecated iwconfig with for example nmcli.
https://connectwww.com/wp-content/uploads/2020/06/nmcli-wifi-details-768x408.png

ghostop14, 2022-11-06:
nmcli won't exist on all linux variants.

chiddekel, 2022-11-06:
> nmcli won't exist on all linux variants
So
R̶e̶p̶l̶a̶c̶e̶ add switch between nmcli and iwconfig

sammyj85, 2023-01-03:
Fedora 36/37 do not have the deprecated iwconfig.
Which distros do not have an nmcli package?

https://www.reddit.com/r/Fedora/comments/w5ughy/iwconfig_not_available_on_fedora/

ghostop14, 2023-01-04:
Raspbian won't for sure.  Ubuntu still has both.

chiddekel, 2023-01-04:
> Raspbian won't for sure. Ubuntu still has both.
If not nmcli then install 

http://www.intellamech.com/RaspberryPi-projects/rpi_nmcli.html

sammyj85, 2023-01-04:
> Raspbian won't for sure. Ubuntu still has both.

The network-manager package looks like it does include nmcli.

https://archive.raspbian.org/raspbian/pool/main/n/network-manager/

chiddekel, 2023-01-05:
> nmcli won't exist on all linux variants.

But can be installed.

sammyj85, 2023-06-29:
iwconfig: "I can still see it in Ubuntu 22.04 but I have been reading about its deprecation for some time now. It’s been [removed from Red Hat](https://access.redhat.com/solutions/1194553?ref=itsfoss.com) and many other distributions.
Suggested alternative: Use the iw command."

https://itsfoss.com/deprecated-linux-commands/

So `iw` may be an alternative as well, which is more widely installed by default.

tazz4843, 2024-10-04:
I just hit this on a Fedora 40 laptop. I can't run sparrow at all because there's zero way for me to install iwconfig anyways.

sammyj85, 2025-03-14:
Ubuntu 24.10 (oracular) has iwconfig in wireless-tools.

https://packages.ubuntu.com/search?suite=noble&arch=any&mode=exactfilename&searchon=contents&keywords=iwconfig

But Ubuntu 25.04 (plucky) does NOT have iwconfig.

https://packages.ubuntu.com/search?suite=plucky&arch=any&mode=exactfilename&searchon=contents&keywords=iwconfig

So this problem is about to become much more prominent.

sammyj85, 2025-10-31:
> > Ubuntu 24.04 (noble) has iwconfig in wireless-tools.
> > https://packages.ubuntu.com/search?suite=noble&arch=any&mode=exactfilename&searchon=contents&keywords=iwconfig
> > But Ubuntu 24.10 (oracular) does NOT have iwconfig.
> > https://packages.ubuntu.com/search?suite=oracular&arch=any&mode=exactfilename&searchon=contents&keywords=iwconfig
> > So this problem is at least now in Ubuntu and Red Hat/Fedora, and becoming more prominent as time passes.
> 
> Look for wireless-tools.

What do you mean - it lists only 22.04 and 24.04 when searching for wireless-tools?

https://packages.ubuntu.com/search?suite=all&arch=any&mode=exactfilename&searchon=names&keywords=wireless-tools

JohnMMcD, 2026-02-22:
If your system has `iw` installed, you can apply this patch to use `iw dev` instead of `iwconfig`. 

[missing_iw_config_try_this.patch](https://github.com/user-attachments/files/25463000/missing_iw_config_try_this.patch)

ghostop14, 2026-02-23:
@JohnMMcD That patch looks pretty straightforward.  If you want to do a PR for that iwconfig/iw section there, I can merge it in.

JohnMMcD, 2026-02-24:
> [@JohnMMcD](https://github.com/JohnMMcD) That patch looks pretty straightforward. If you want to do a PR for that iwconfig/iw section there, I can merge it in.

Great - added https://github.com/ghostop14/sparrow-wifi/pull/115 . I added support for `nmcli`, too.

ghostop14, 2026-04-26:
Added fallback iw support where iwconfig was used.