ghostop14/sparrow-wifi #81: Program crash when accessing telemetry for a targeted network

roseninc, 2022-11-16:
Using the latest version, I get a crash whenever I try to access telemetry. Everything else is working as perfectly fine.

The error I get:

/home/kali/sparrow-wifi/telemetry.py:92: UserWarning: FixedFormatter should only be used together with FixedLocator
  self.axes.set_yticklabels(['-20', '-40', '-60', '-80', '-100'], color=self.fontColor)
Traceback (most recent call last):
  File "/home/kali/sparrow-wifi/telemetry.py", line 225, in resizeEvent
    self.radar.setGeometry(self.geometry().width() - smallerDim - 10, 10, smallerDim, smallerDim)
TypeError: arguments did not match any overloaded call:
  setGeometry(self, QRect): argument 1 has unexpected type 'float'
  setGeometry(self, int, int, int, int): argument 1 has unexpected type 'float'
zsh: IOT instruction  ./sparrow-wifi.py

ghostop14, 2022-11-18:
Yeah the newer versions of python haven't been making the type conversions correctly anymore.  So these keep popping up.  I just pushed an update that looks like it fixes it.  Do a fresh pull and let me know if that takes care of it for you (worked here)

roseninc, 2022-11-24:
> Yeah the newer versions of python haven't been making the type conversions correctly anymore. So these keep popping up. I just pushed an update that looks like it fixes it. Do a fresh pull and let me know if that takes care of it for you (worked here)

Can confirm that it works now! Thanks for creating such an awesome tool. It's perfect for my pentesting assesments :D

closed 2022-11-24 by roseninc

alphafox02, 2024-05-29:
Fast forward to 2024 - with the latest git pull on 22.04 I see the following when closing at Sparrow-Wifi.

/usr/src/sparrow-wifi/telemetry.py:92: UserWarning: FixedFormatter should only be used together with FixedLocator
  self.axes.set_yticklabels(['-20', '-40', '-60', '-80', '-100'], color=self.fontColor)