ghostop14/sparrow-wifi #82: rtl8812au issue/support?

i486, 2022-12-18:
**Issue: sparrow-wifi doesn't show anything after starting the scan**

**Wi-Fi driver**: https://github.com/aircrack-ng/rtl8812au
**OS**: Kali 2022.4
```
sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get

.0.0-kali5-amd64 x86_64
EFI variables are not supported on this system
Bus 001 Device 004: ID 0b05:17d2 ASUSTek Computer, Inc. USB-AC56 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
Bus 001 Device 005: ID 0b05:17d2 ASUSTek Computer, Inc. USB-AC56 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
2: phy2: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: phy3: Wireless LAN
        Soft blocked: no
        Hard blocked: no
8812au/5.6.4.2_35491.20191025, 6.0.0-kali5-amd64, x86_64: installed
phy#3
        Interface wlan0
                ifindex 6
                wdev 0x300000001
                addr f8:32:e4:b5:ed:0b
                type monitor
                txpower 20.00 dBm
phy#2
        Interface wlan1
                ifindex 5
                wdev 0x200000001
                addr f8:32:e4:b3:b6:e2
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 20.00 dBm
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#3
country 99: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 40), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5140 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN

phy#2
country 99: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 40), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5140 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN

```

ghostop14, 2022-12-19:
Depending on the OS, the stock 8812 drivers can be problematic.  Generally I manually build/install the one that's available with aircrack-ng and that does the trick:  https://github.com/aircrack-ng/rtl8812au.git

ghostop14, 2022-12-19:
From the command-line with the interface you're using, see what this command outputs:  

sudo iw dev <interface name> scan