ghostop14/sparrow-wifi #89: Cannot install some requiremenets.

tsalisbury0, 2023-05-23:
Pip3 is giving me errors when trying to install.

**error**: externally-managed-environment

**×** This environment is externally managed
**╰─>** To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

yarcod, 2023-09-19:
This is due to a change in how global Python packages are handled from Debian 12 and Ubuntu 23.04. One of the reasons for this is for users and distribution maintainers to have less trouble with people's Python environments being screwed up by some package installation. The solution is what has been common practice for a while, anyway: Virtual Environments (venv), which isolates package installs to a limited scope of your system. With that said, you could "work around" this and accomplish what `pip3` used to do here by using `pipx` instead.

As per the output you posted, you can enable a virtual environment with `python3 -m venv  /some/path`. However I would recommend you to check out any of the number of wrappers that makes it a lot easier to deal with virtual environments. I personally use `pipenv`, along with `pipenv-pipes` to easily navigate and switch venv as needed.

fernandoroa, 2024-05-17:
The `sudo` command in the README, for the virtual environment case should be:  
`sudo -E env PATH=${PATH} python sparrow-wifi.py`