ghostop14/sparrow-wifi #91: CatSniffer support

alphafox02, 2023-08-12:
I came across this at DefCon and thought maybe it’d be something to incorporate into Sparrow as the Ubertooth becomes harder to get.

https://github.com/ElectronicCats/CatSniffer

https://github.com/ElectronicCats/Sniffle

ghostop14, 2023-08-14:
I saw the names coming in the DefCon social media posts but hadn't had a chance to look into them yet.  Definitely interesting.

alphafox02, 2023-10-29:
I finally have the sniffle firmware running on a CatSniffer 3.1

https://github.com/nccgroup/Sniffle

Not sure how applicable this is, but 

“PCAP export compatible with the Ubertooth”

Still can’t help to think maybe it’d be a great tool to feed Sparrow’s bluetooth portions.