ghostop14/sparrow-wifi #92: sparrow-wifi.py  not executing in terminal

Noob-hacker369, 2023-11-21:
Traceback (most recent call last):
  File "/home/kali/tools/sparrow-wifi/./sparrow-wifi.py", line 37, in <module>
    from PyQt5.QtChart import QChart, QChartView, QLineSeries, QValueAxis
ImportError: /usr/local/lib/python3.11/dist-packages/PyQt5/Qt5/lib/libQt5Charts.so.5: undefined symbol: _ZdlPvm, version Qt_5

showing this problem

Bert-H, 2023-12-20:
I have the exact same error on Fedora38. I had not used Sparrow-wifi for maybe about a year and it worked fine the last time.
Now when starting it, it comes up with the same error as mentioned above.
I followed the instructions in the README.md file again, upgraded pip with "sudo pip3 install --upgrade pip" which did upgrade some components but still the same error. Also re-installed other modules as per the instructions, but no fix.
Noticed that FC38 has very little support it seems for qt5? Almost all references are for qt6 such as 'python3-pyqt6-charts".
Is this the issue? If so, how to fix it?

Bert-H, 2023-12-21:
OK, fixed it using https://github.com/ghostop14/sparrow-wifi/issues/1 as a guide.

Noob-hacker369, 2023-12-22:
My issue now resolved. 
by the commands :-
sudo pip3 install --upgrade pip
sudo apt-get install --reinstall python3-pyqt5
pip3 install --upgrade --force-reinstall PyQtChart

closed 2023-12-22 by Noob-hacker369