ghostop14/sparrow-wifi #97: SyntaxWarning with Python 3.12

elboulangero, 2024-07-25:
With Python 3.12:

```
/usr/share/sparrow-wifi/sparrowwifiagent.py:285: SyntaxWarning: invalid escape sequence '\.'
  ippattern = re.compile('([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3})')
```

This change is documented at https://docs.python.org/3/whatsnew/3.12.html#other-language-changes:

> A backslash-character pair that is not a valid escape sequence now generates a [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning), instead of [DeprecationWarning](https://docs.python.org/3/library/exceptions.html#DeprecationWarning). For example, re.compile("\d+\.\d+") now emits a [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning) ("\d" is an invalid escape sequence, use raw strings for regular expression: re.compile(r"\d+\.\d+")). In a future Python version, [SyntaxError](https://docs.python.org/3/library/exceptions.html#SyntaxError) will eventually be raised, instead of [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning).

elboulangero, 2024-07-25:
Tentative patch:

```
From ead55fc5007199e6cae67cca5736b93c00dcaf18 Mon Sep 17 00:00:00 2001
From: Arnaud Rebillout <arnaudr@kali.org>
Date: Thu, 25 Jul 2024 15:42:10 +0200
Subject: [PATCH] Fix Python 3.12 Syntax Warnings (use raw strings)

---
 sparrow-wifi.py     | 6 +++---
 sparrowdialogs.py   | 2 +-
 sparrowwifiagent.py | 2 +-
 3 files changed, 5 insertions(+), 5 deletions(-)

diff --git a/sparrow-wifi.py b/sparrow-wifi.py
index 38c8ba1..8fd9877 100755
--- a/sparrow-wifi.py
+++ b/sparrow-wifi.py
@@ -1796,7 +1796,7 @@ class mainWindow(QMainWindow):
         text, okPressed = QInputDialog.getText(self, "Remote Agent","Please provide gpsd IP:", QLineEdit.Normal, "127.0.0.1:2947")
         if okPressed and text != '':
             # Validate the input
-            p = re.compile('^([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}:[0-9]{1,5})')
+            p = re.compile(r'^([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}:[0-9]{1,5})')
             specIsGood = True
             try:
                 agentSpec = p.search(text).group(1)
@@ -3346,7 +3346,7 @@ class mainWindow(QMainWindow):
                 return False, "", 0
             
             # Validate the input
-            p = re.compile('^([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}:[0-9]{1,5})')
+            p = re.compile(r'^([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}:[0-9]{1,5})')
             specIsGood = True
             try:
                 agentSpec = p.search(text).group(1)
@@ -3443,7 +3443,7 @@ class mainWindow(QMainWindow):
             text, okPressed = QInputDialog.getText(self, "Remote Agent","Please provide the <IP>:<port> of the remote agent\nor specify 'auto' to launch agent listener\n(auto requires agent to be on the same subnet and started with the --sendannounce flag):", QLineEdit.Normal, "127.0.0.1:8020")
             if okPressed and text != '':
                 # Validate the input
-                p = re.compile('^([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}:[0-9]{1,5})')
+                p = re.compile(r'^([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}:[0-9]{1,5})')
                 specIsGood = True
                 try:
                     agentSpec = p.search(text).group(1)
diff --git a/sparrowdialogs.py b/sparrowdialogs.py
index 18cbee7..c77b833 100755
--- a/sparrowdialogs.py
+++ b/sparrowdialogs.py
@@ -1974,7 +1974,7 @@ class AgentConfigDialog(QDialog):
 
     def validateAllowedIPs(self, allowedIPstr):
         if len(allowedIPstr) > 0:
-            ippattern = re.compile('([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3})')
+            ippattern = re.compile(r'([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3})')
             if ',' in allowedIPstr:
                 tmpList = allowedIPstr.split(',')
                 for curItem in tmpList:
diff --git a/sparrowwifiagent.py b/sparrowwifiagent.py
index 4b49df9..51e5f78 100755
--- a/sparrowwifiagent.py
+++ b/sparrowwifiagent.py
@@ -282,7 +282,7 @@ def buildAllowedIPs(allowedIPstr):
     allowedIPs = []
 
     if len(allowedIPstr) > 0:
-        ippattern = re.compile('([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3})')
+        ippattern = re.compile(r'([0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3})')
         if ',' in allowedIPstr:
             tmpList = allowedIPstr.split(',')
             for curItem in tmpList:
-- 
2.45.2

```

ghostop14, 2024-07-25:
Thanks!  I'll test it out and integrate the changes.

ghostop14, 2024-07-25:
Just pushed the updates.

closed 2024-07-25 by ghostop14