ghostop14/sparrow-wifi #106: No Module Named 'gps3' when running as root

donwerder, 2025-07-31:
A scan window opens but states that I need to be root to run a scan. When I run as root (sudo) it states it can’t find gps3.

(sparrow) don@Ubuntu-MacBookPro:~/sparrow-wifi$ python3 sparrow-wifi.py
/home/don/sparrow-wifi/sparrowbluetooth.py:369: SyntaxWarning: invalid escape sequence '\('
  p_company = re.compile('Company: (.*) \(')
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
You need to have root privileges to run this script.
Please try again, this time using 'sudo'. Exiting.
(sparrow) don@Ubuntu-MacBookPro:~/sparrow-wifi$ sudo python3 sparrow-wifi.py
Traceback (most recent call last):
  File "/home/don/sparrow-wifi/sparrow-wifi.py", line 45, in <module>
    from wirelessengine import WirelessEngine, WirelessNetwork
  File "/home/don/sparrow-wifi/wirelessengine.py", line 29, in <module>
    from sparrowgps import SparrowGPS
  File "/home/don/sparrow-wifi/sparrowgps.py", line 27, in <module>
    from gps3.agps3threaded import AGPS3mechanism
ModuleNotFoundError: No module named 'gps3'

rikosintie, 2025-09-18:
After you start the python virtual environment:

Try "sudo path/to/venv/bin/python3 sparrow-wifi.py"

When you use sudo it calls the system python, not the virtual environment python. 

Use "which python3" to see the executable 

Use "which -a python3" to see all executables

alphafox02, 2025-10-31:
Add the following to your pip command --break-system-packages